rust-embedded/svd2rust #20: An alternative write/modify interface

phil-opp, 2016-11-01:
Thanks for the great library!

I have some problems with the generated modify function. It leads to very verbose code when we want to change a single field. For example, let's assume that we want to change the mode of GPIO A pin 0 to 1:

```rust
gpioa.moder.modify(|r,w| w.moder0(1).moder1(r.moder1()).moder2(r.moder2())
    .moder3(r.moder3()).[…].moder15(r.moder15()));
```

Why not generate a modify function like that:

```rust
gpioa.moder.modify(|m| m.moder0(1));
```

The `m` would be a write instance that has all values set to the current value (instead of the reset value).

With such a modify function, we could replace the `write` function with a combination of `modify` and a new `reset` function. So instead of:

```rust
gpioa.moder.write(|w| w.moder0(0));
```

we would write

```rust
gpioa.moder.reset().modify(|m| m.moder0(0));
```

Which – in my opinion – expresses the intent in a clearer way.

japaric, 2016-11-01:
> Thanks for the great library!

You are welcome.

> For example, let's assume that we want to change the mode of GPIO A pin 0 to 1:

That would be just:

``` rust
// Change (only) the mode of PA0
gpioa.moder.modify(|_, w| w.moder0(1));
```

`w` already contains the current state, not the reset value. The reset value is only used in the `write` method. Could this be a bug in the documentation? Perhaps, it needs more examples.

"`r`" is mainly used to *toggle* the value of some bit or express some other conditional change:

``` rust
// Toggle PA0
gpioa.odr(|r, w| w.odr0(!r.odr0()));
```

because you can't access the bits of `ODR` through `w`.

> gpioa.moder.reset().modify(|m| m.moder0(0));

I like how this reads. If we can somehow make it compile to a *single* volatile write then I'd be happy to make this change.

Thanks for the feedback! I appreciate it.

phil-opp, 2016-11-01:
Thanks for the quick reply.

> w already contains the current state, not the reset value.

I see. The additional r parameter confused me (but I understand it now) and the documentation is not really clear about this. I agree that more examples would be useful (e.g. an example without any toggle).

> … If we can somehow make it compile to a _single_ volatile write then I'd be happy to make this change.

Hmm, I don't have any idea how this could work right now… But I'll come back at you if I find a solution.

closed 2016-11-01 by phil-opp