rust-embedded/svd2rust #23: Odd formatting for output

thejpster, 2016-11-05:
I build svd2rust, using rust 1.12.1, with

```bash
cargo install svd2rust
```

If I run it with the Kinetis MKE06Z4.svd file, I get this output

```rust
impl PidrW { # [ doc = r" Reset value" ] pub fn reset_value ( ) -> Self { PidrW { bits : 4294967295u32 } } # [ doc = "Bits 0:31 - Port Input Disable" ] pub fn pid ( & mut self , value : u32 ) -> & mut Self { const OFFSET : u8 = 0u8 ; const MASK : u32 = 4294967295 ; self . bits &= ! ( ( MASK as u32 ) << OFFSET ) ; self . bits |= ( ( value & MASK ) as u32 ) << OFFSET ; self } }
```

I can fix the #[] pragmas manually and then rustfmt it back into shape, but it's a bit weird.

OS is Ubuntu 14.04 for x86-64.

japaric, 2016-11-05:
> I get this output

Yeah, that's what the `quote` crate generates; it doesn't preserve the format of the templates we feed to it. `rustfmt`ing the output makes it readable though.

> I can fix the #[] pragmas

By fix, you mean that the code doesn't compile if not modified?

thejpster, 2016-11-05:
I had assumed that rustfmt wouldn't like it if the #[] attributes weren't at the start of line, but my mistake, it's quite happy. However, rustfmt doesn't manage to remove the whitespace, so that needs fixing by hand. It compiles fine, so I guess this is just a matter of style.

japaric, 2017-04-10:
@JayKickliter Sounds like you should report that to the rustfmt devs.

A possible workaround (I haven't tried it myself) might be using `tr` to replace spaces with newlines before feeding the crate to rustfmt. (Assuming that having a single very long line is the problem)

JayKickliter, 2017-04-11:
Thanks. I tried your suggesting to use `tr` and it looks like there's also some fundamental problem with the generated code. I'll open an issue if I continue down this route.

japaric, 2017-04-11:
@JayKickliter The code should compile unformatted. If it doesn't, that's a bug and I would appreciate a bug report 😄. Just pointing me to the SVD file would suffice.

japaric, 2017-04-15:
This unformatted output is expected and the [docs](https://docs.rs/svd2rust/0.6.0/svd2rust/) recommend piping svd2rust output before writing it to a file.

@JayKickliter Please open an issue if you find that the generated code doesn't compile!

closed 2017-04-15 by japaric