rust-embedded/svd2rust #27: Compilation fails on generated file with '%s'

lexxvir, 2016-11-18:
I've tried to generate *.rs from [MK70F15.xml.txt](https://github.com/japaric/svd2rust/files/599267/MK70F15.xml.txt) (I added 'txt' extension to satisfy github, I believe it is SVD file, I found it [here](https://sourceforge.net/projects/mcuoneclipse/files/Eclipse%20Plugins/EmbSysReg/)).

Output file generated successfully (some warnings though):
```
svd2rust spi0 -i ./MK70F15.xml > src/spi0.rs
WARNING CTAR_SLAVE overlaps with another register at offset 12. Ignoring.
WARNING PUSHR_SLAVE overlaps with another register at offset 52. Ignoring.
```

But following error appeared on compilation:
```
error: expected `:`, found `%`
 --> src/spi0.rs:3:295
  |
3 | # [ repr ( C ) ] pub struct Spi0 { # [ doc = "0x00 - DSPI Module Configuration Register" ] pub mcr : Mcr , _reserved0 : [ u8 ; 4usize ] , # [ doc = "0x08 - DSPI Transfer Count Register" ] pub tcr : Tcr , # [ doc = "0x0c - DSPI Clock and Transfer Attributes Register (In Master Mode)" ] pub ctar%s : Ctar%s , _reserved1 : [ u8 ; 28usize ] , # [ doc = "0x2c - DSPI Status Register" ] pub sr : Sr , # [ doc = "0x30 - DSPI DMA/Interrupt Request Select and Enable Register" ] pub rser : Rser , # [ doc = "0x34 - DSPI PUSH TX FIFO Register In Master Mode" ] pub pushr : Pushr , # [ doc = "0x38 - DSPI POP RX FIFO Register" ] pub popr : Popr , # [ doc = "0x3c - DSPI Transmit FIFO Registers" ] pub txfr%s : Txfr%s , _reserved2 : [ u8 ; 60usize ] , # [ doc = "0x7c - DSPI Receive FIFO Registers" ] pub rxfr%s : Rxfr%s }
  |
```

Probably svd2rust should somehow convert '%s' in SVD into something legal for Rust.

japaric, 2016-11-27:
That SVD file contains stuff like this:

``` xml
        <register>
          <dim>8</dim>
          <dimIncrement>0x100</dimIncrement>
          <dimIndex>0,1,2,3,4,5,6,7</dimIndex>
          <name>PRS%s</name>
```

which represents an array of registers. So svd2rust should actually expand that to something like this:

``` rust
struct Axbs {
    prs: Prs[8],
}

struct Prs {
    register: u32,
    padding: [u8; 224],
}

impl Prs { /* read, write, modify methods */ }
```

but that hasn't been implemented yet. Actually, I just realized that sort of array encoding exists 😄.

istankovic, 2016-12-10:
@japaric shouldn't this be 
```rust
struct Axbs {
    prs: [Prs; 8],
}

struct Prs {
    register: u32,
    padding: [u8; 252],
}
```
? Note the padding (dimIncrement is specified in bytes whereas the register size is in bits).

japaric, 2016-12-10:
@istankovic Possibly, I didn't actually look at the units.

japaric, 2017-04-15:
This was implemented in #43

closed 2017-04-15 by japaric