rust-embedded/svd2rust #34: Optimize access to bitfields

japaric, 2016-11-27:
For instance the I2C peripheral of the STM32F303 has a "32-bit" TXDR register that only contains 8 accessible bits (the other bits are reserved). In that case, svd2rust generates code like this:

``` rust
# [ repr ( C ) ]
pub struct Txdr {
    register: ::volatile_register::RW<u32>,
}

impl Txdr {
    // ...

    pub fn write<F>(&mut self, f: F)
        where F: FnOnce(&mut TxdrW) -> &mut TxdrW
    {
        let mut w = TxdrW::reset_value();
        f(&mut w);
        self.register.write(w.bits);
    }
}

# [ derive ( Clone , Copy ) ]
# [ repr ( C ) ]
pub struct TxdrW {
    bits: u32,
}

impl TxdrW {
    # [ doc = r" Reset value" ]
    pub fn reset_value() -> Self {
        TxdrW { bits: 0u32 }
    }
    # [ doc = "Bits 0:7 - 8-bit transmit data" ]
    pub fn txdata(&mut self, value: u8) -> &mut Self {
        const OFFSET: u8 = 0u8;
        const MASK: u8 = 255;
        self.bits &= !((MASK as u32) << OFFSET);
        self.bits |= ((value & MASK) as u32) << OFFSET;
        self
    }
}
```

Which is wasteful because it performs 32-bit loads/stores when it could be doing 8-bit loads/stores. Instead svd2rust should generate code like this:

``` rust
# [ repr ( C ) ]
pub struct Txdr {
    register: ::volatile_register::RW<u8>,
    padding: [u8; 3],
}

impl Txdr {
    // ...

    pub fn write<F>(&mut self, f: F)
        where F: FnOnce(&mut TxdrW) -> &mut TxdrW
    {
        let mut w = TxdrW::reset_value();
        f(&mut w);
        self.register.write(w.bits);
    }
}

# [ derive ( Clone , Copy ) ]
# [ repr ( C ) ]
pub struct TxdrW {
    bits: u8,
}

impl TxdrW {
    # [ doc = r" Reset value" ]
    pub fn reset_value() -> Self {
        TxdrW { bits: 0u32 }
    }
    # [ doc = "Bits 0:7 - 8-bit transmit data" ]
    pub fn txdata(&mut self, value: u8) -> &mut Self {
        self.bits = value;
        self
    }
}
```

This can be changed without affecting the generated API.

whitequark, 2017-04-29:
> Which is wasteful because it performs 32-bit loads/stores when it could be doing 8-bit loads/stores.

I'm pretty sure this is not legal. I have actually had a C compiler emit narrower than they should loads/stores (because I had the bright idea of using bitfields... I was young and stupid), and some peripheral on STM32 (I think it was the GPIO one?) just did nothing in response to my writes. IIRC it is explicitly stated in the reference manual somewhere.

japaric, 2017-04-29:
> IIRC it is explicitly stated in the reference manual somewhere.

For the particular example in the issue description, the reference manual says

> The peripheral registers are accessed by words (32-bit)

But this particular text is at the beginning of the registers section rather than repeat on each register description so it's easy to miss.

> I'm pretty sure this is not legal

Yes. It's not legal in the general case. Though I have seen, at least once, a 16-bit (timer, I think?) register that was access as a word because that's how the SVD described it even though the reference manual said it *could* be accessed as a byte, half-word or word. But that could be considered a missing feature of SVD files; they don't have a way to encode the idea of "multiple access pattens". So I think this should be optimized on a case basis and in the SVD file rather than in svd2rust. So closing.

closed 2017-04-29 by japaric