rust-embedded/svd2rust #40: [Feature] Support interrupt table generation

grossws, 2016-12-15:
SVD files usually contain interrupts which can be used to generate interrupt vector table. 
Maybe support two modes:
- generate full table (both events and interrupts),
- generate only array with interrupts.

I'm not sure how it should look like, so this issue is just to fix absence of such feature. If I get some spare time, I'll try to experiment with it.

japaric, 2016-12-16:
> I'm not sure how it should look like

This is the main issue. I'm currently "declaring" overridable interrupts like [this](https://github.com/japaric/f3/blob/v0.3.0/src/interrupt.rs#L16-L254) (plus [this](https://github.com/japaric/f3/blob/v0.3.0/build.rs#L102-L176)) but I don't know if everyone else wants to do it the same way ...

> generate full table (both events and interrupts),

By "events" you mean *exceptions* (the part of the vector table that comes before the actual interrupts)?

grossws, 2016-12-16:
> By "events" you mean exceptions (the part of the vector table that comes before the actual interrupts)? 

Yes, I thought about exceptions + interrupts.

commit 62a58d11 (2017-03-12): whole device generation

Fixes #60
Closes #40
PR #61 (2017-03-12): whole device generation
Fixes #60
Closes #40
closed 2017-03-12 by japaric