rust-embedded/svd2rust #41: Make `bits` pub?

astro, 2016-12-20:
I've got 4 LEDs that I can manipulate individually. My first try looks like this:
```rust
fn toggle_led(led: u8) {
    match led {
        1 => 
            gpio_port().not[2].write(|notw| notw.notp1(true)),
        2 => 
            gpio_port().not[2].write(|notw| notw.notp2(true)),
        3 => 
            gpio_port().not[2].write(|notw| notw.notp8(true)),
        4 => 
            gpio_port().not[5].write(|notw| notw.notp26(true)),
        _ => (),
    }
}
```

It's hard to parameterize over `notp1`, `notp2`, etc. In this case I'd like to sacrifice the pretty abstraction and manipulate any bit of `self.bits`.

japaric, 2016-12-21:
We can add an unsafe accessor to get `&mut bits`.

japaric, 2016-12-21:
We can make `bits` public because that would make it "safe" to modify `bits` in a way that breaks the invariants that the abstraction preserves.

commit 7457cf37 (2016-12-31): expose bits

closes #41
closed 2016-12-31 by homunkulus