rust-embedded/svd2rust #48: Add Debug trait implementation

newpavlov, 2016-12-27:
It would be very helpful to have a `Debug` traits implemented for register structs. Currently if I am debugging some code or interested in how things work I simply add something like [this](https://gist.github.com/anonymous/6da8dd5b9dabce394ce9f8a7939b3c84) to memory map crate. It will be mainly useful for read and memory block structs, but some may find it usefull for write structs too.

japaric, 2016-12-27:
> It would be very helpful to have a Debug traits implemented for register structs.

Are you serializing/printing the registers through UART/ITM? I wonder if this could be handled with GDB pretty printers, then:

```
(gdb) print block.register
{
    fooen: true,
    baren: false,
    ..
}
```

would Just Work.

In any case, I'm OK with adding `Debug` implementations to `RegisterR`s. The implementation should use the ["formatting structs"](https://doc.rust-lang.org/std/fmt/struct.Formatter.html#method.debug_struct) though, then we get pretty printing (`{#?}`) for free.

newpavlov, 2016-12-27:
I am using ITM. If I am not mistaken this will not work nicely with projects compiled with `--release` (at least with default options). Also it will be friendlier for user to just use several `iprinln!`'s to see how registers change without any additional typing in the console.

"formatting structs" will be a great way to do it.

hannobraun, 2020-07-22:
I assume this issue refers to the `RegisterBlock` structs. In addition, it would be useful to also have `Debug` implementations for the peripheral structs that dereference to them (the one's with the `ptr` method).

burrbull, 2023-11-24:
Fixed by #716

closed 2023-11-24 by burrbull