rust-embedded/svd2rust #51: Better error messages with parse error.

Emilgardis, 2017-01-08:
I'm trying to read the svd for nuvoton M451 found in here `http://www.nuvoton.com/hq/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicro_DFP.1.0.9.pack` (Note, don't download via browser, use `wget`).
The error I get is this:
```text
$ RUST_BACKTRACE=1 svd2rust -i M451_v1.svd 
thread 'main' panicked at '.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.2.0/src/lib.rs:185 tree.get_child_text("description")', /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libcore/option.rs:715
stack backtrace:
   1:     0x55c1576c20ea - std::sys::imp::backtrace::tracing::imp::write::h0d1aacfb8fc693ac
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:42
   2:     0x55c1576c45cf - std::panicking::default_hook::{{closure}}::hff87359baf5754d0
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/panicking.rs:349
   3:     0x55c1576c41ce - std::panicking::default_hook::h936f17ca3b2b98dc
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/panicking.rs:365
   4:     0x55c1576c4a17 - std::panicking::rust_panic_with_hook::he9c830ab830c7944
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/panicking.rs:553
   5:     0x55c1576c48a4 - std::panicking::begin_panic::hbd10b7f500042f98
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/panicking.rs:515
   6:     0x55c1576c47c9 - std::panicking::begin_panic_fmt::h49583ddd44614a0d
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/panicking.rs:499
   7:     0x55c1576c4757 - rust_begin_unwind
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/panicking.rs:475
   8:     0x55c1576ecb7d - core::panicking::panic_fmt::h3b787d0ceafd7204
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libcore/panicking.rs:69
   9:     0x55c1576ecbed - core::option::expect_failed::hfd8d57ff225ddc33
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libcore/option.rs:715
  10:     0x55c1576a28b2 - svd_parser::Register::parse::h5423bbeb0d22c01a
  11:     0x55c1576a1790 - svd_parser::Device::parse::hbaabf4361b1d1096
  12:     0x55c1576a0868 - svd_parser::parse::h13d5d8c2807f67bc
  13:     0x55c15762bc97 - svd2rust::main::hb6cdc518cc3a4731
  14:     0x55c1576cb8aa - __rust_maybe_catch_panic
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libpanic_unwind/lib.rs:98
  15:     0x55c1576c5156 - std::rt::lang_start::ha749a62502c90792
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/panicking.rs:434
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/panic.rs:351
                        at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/libstd/rt.rs:57
  16:     0x7fdbf8a9c290 - __libc_start_main
  17:     0x55c157628a09 - _start
  18:                0x0 - <unknown>
```
Which is very unhelpful. If `svd-parser` and `svd2rust` used `error_chain` I think this can be fixed.
The source of this error though is wierd though. I've also opened an issue in `svd-parser` to highlight the current `try!` macro defined inside it.

Emilgardis, 2017-04-28:
See japaric/svd#26 for work on the pr,

adamgreig, 2019-07-26:
Error messages are now improved (especially thanks to #237 and #317), and this issue is years old, please re-open if required.

closed 2019-07-26 by adamgreig

nickray, 2019-08-14:
Opened https://github.com/rust-embedded/svd2rust/issues/376 as I can't reopen this one.