rust-embedded/svd2rust #54: Rust keywords are not escaped

crawford, 2017-02-05:
I ran svd2rust over the DMA section of the EFM32LG990F256 and got the following code:

```rust
# [ doc = "DMA" ]                                                                                                                                                                         
# [ repr ( C ) ]
pub struct Dma {
    ...
    pub if: If,
    # [ doc = "0x1004 - Interrupt Flag Set Register" ]
    pub ifs: Ifs,                                                                                                                                                                         
    # [ doc = "0x1008 - Interrupt Flag Clear Register" ]
    pub ifc: Ifc,
    # [ doc = "0x100c - Interrupt Enable register" ]
    ...
```

Unfortunately, this does not compile because `if` is a reserved keyword. I'd like to help out on the solution for this, but I'm not sure which direction you'd like to go. If you were to add a suffix or prefix to every register and field, you wouldn't have to do any special escaping and you wouldn't have special-cased names which don't follow the SVD. For example:

```rust
# [ doc = "DMA" ]                                                                                                                                                                         
# [ repr ( C ) ]
pub struct PeripheralDma {
    ...
    pub reg_if: If,
    # [ doc = "0x1004 - Interrupt Flag Set Register" ]
    pub reg_ifs: Ifs,                                                                                                                                                                         
    # [ doc = "0x1008 - Interrupt Flag Clear Register" ]
    pub reg_ifc: Ifc,
    # [ doc = "0x100c - Interrupt Enable register" ]
    ...
```

The downside to that approach, is that you now have lots of repetition which hurts readability.

Emilgardis, 2017-02-05:
I've gotten the same thing, there is a sanitizer (https://github.com/japaric/svd2rust/blob/master/src/lib.rs#L267) but it doesn't have `if`.

crawford, 2017-02-05:
@Emilgardis I just stumbled upon that as well. I wrote a [patch](https://github.com/japaric/svd2rust/pull/55) to use the full list.

closed 2017-02-09 by homunkulus