rust-embedded/svd2rust #57: Add option to use different name for generated struct

prusnak, 2017-02-12:
See the followings script for example:

https://github.com/brandonedens/stm32l4x6-memory-map/blob/master/gen-peripherals.sh

It uses `sed` to rename struct name from `I2c1` to `I2c` if the definition could be shared among more peripherals of the same type.

It would be nice to have option directly in svd2rust so one does not need to use sed.

For example `svd2rust -i $svd i2c1:I2c ` or `svd2rust -i $svd i2c1 -n I2c`

japaric, 2017-04-15:
Today, we (a) generate code for the whole device instead of per peripheral, (b) generate one type per peripheral instance and (c) the name of that type matches the peripheral instance (`I2c1`, `I2c2`, etc.) so it's no longer necessary to (a) use a shell script to build a device crate or (b) sed the output.

I feel that this no longer required so I'm going to close this issue. If you feel otherwise, feel free to open a new issue with updated rationale.

closed 2017-04-15 by japaric

prusnak, 2017-04-15:
Great! Thanks!