rust-embedded/svd2rust #60: Enum names collisions inside a register

jboone, 2017-03-05:
Thanks @japaric for the amazing volume and quality of embedded work you're doing for the Rust community.

I'm working on [rust-lpc43xx](https://github.com/astro/rust-lpc43xx), and the [LPC43xx SVD](https://www.lpcware.com/content/nxpfile/lpc43sxx-svd-file) describes several fields in a register using the same \<enumeratedValues\> name, "ENUM". See for instance Timer0 peripheral's fields MR0I and MR0R. There are a dozen or so other examples for Timer0 alone, and a quick scan of the SVD indicates they use "ENUM" for *all* their \<enumeratedValues\> -- hundreds of examples.

I can think of a few ways to address this:

1. Name the enum struct using a concatenation of the field name and enumeratedValues name. This could result in longer names when referencing the enum, with the field name repeated in the enum struct name.
2. Find a way to scope the enum inside the field struct, or some other namespace. This seems a bit tidier, but not sure you can declare an enum inside a struct?
3. Ignore the enumeratedValues name altogether, if it's common practice for it to always be "ENUM", or if a field can only have one \<enumeratedValues\> block. I can't speak to whether those assumptions are true or not...

I think I understand the svd2rust code well enough that I could attempt to make any of the three changes outlined above. But I'm hoping somebody with a more informed opinion can speak up and help decide which is the best choice.

japaric, 2017-03-06:
> 3. Ignore the enumeratedValues name altogether

I'm leaning towards this option.

---

Right now we use `enumeratedValues.name` for the enum name to get less worse names.

For example, given the following SVD file (modified [nrf51.svd](https://github.com/posborne/cmsis-svd/blob/aa4721af946a253d18c8737b01d23e9c88a42e84/data/Nordic/nrf51.svd))

``` xml
      <registers>
        <register>
          <name>OUT</name>
          <description>Write GPIO port.</description>
          <addressOffset>0x504</addressOffset>
          <fields>
            <field>
              <name>PIN0</name>
              <description>Pin 0.</description>
              <lsb>0</lsb> <msb>0</msb>
              <enumeratedValues>
                <name>PIN</name>
                <enumeratedValue>
                  <name>Low</name>
                  <description>Pin driver is low.</description>
                  <value>0</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>High</name>
                  <description>Pin driver is high.</description>
                  <value>1</value>
                </enumeratedValue>
              </enumeratedValues>
            </field>
            <field>
              <name>PIN1</name>
              <description>Pin 1.</description>
              <lsb>1</lsb> <msb>1</msb>
              <enumeratedValues derivedFrom="PIN">
              </enumeratedValues>
```

We generate something like this:

``` rust
pub mod out {
    pub struct R { .. }

    pub enum PinR { .. }

    impl R {
        pub fn pin0(&self) -> PinR { .. }
        pub fn pin1(&self) -> PinR { .. }
        ..
    }
}
```

But if we used `field.name` instead of `enumeratedValues.name` then we would end with something like this:

``` rust
pub mod out {
    pub struct R { .. }

    pub enum Pin0R { .. }

    impl R {
        pub fn pin0(&self) -> Pin0R { .. }
        pub fn pin1(&self) -> Pin0R { .. }
        ..
    }
}
```

Which looks weird: `pin1` returns `Pin0R`.

---

The change I'm picturing is the following:

We use `field.name` instead of `enumeratedValues.name` *but* we create type aliases when dealing with `derivedFrom` in `enumeratedValues`.

So we should generate something like this:

``` rust
pub mod out {
    pub struct R { .. }

    pub enum Pin0R { .. }

    pub type Pin1R = Pin0R;

    impl R {
        pub fn pin0(&self) -> Pin0R { .. }
        pub fn pin1(&self) -> Pin1R { .. }
        ..
    }
}
```

cc @perlindgren

jboone, 2017-03-07:
Using field.name instead of enumeratedValue.name change is made here: https://github.com/jboone/svd2rust/commit/57d298ac886b3649ebb418773dcfd088c8238fff

Type aliasing is a bigger piece of work, and I don't have a very good feel for the code yet...

commit 62a58d11 (2017-03-12): whole device generation

Fixes #60
Closes #40
PR #61 (2017-03-12): whole device generation
Fixes #60
Closes #40
closed 2017-03-12 by japaric