rust-embedded/svd2rust #68: v0.5.1 Output invalid and all on one line with STM32F41x.svd

archaelus, 2017-04-02:
I ran `rustup run nightly cargo install svd2rust` a couple of days ago using `rustc 1.18.0-nightly (40feadb96 2017-03-31)` and then ran `svd2rust -i STM32F41x.svd > stm32f415.rs` with [STM32F41x.svd](https://raw.githubusercontent.com/posborne/cmsis-svd/master/data/STMicro/STM32F41x.svd).

The output was a rust file with all the code on one line:
```
$ wc stm32f415.rs
       1 2284030 7837650 stm32f415.rs
```

Trying to `rustfmt` the `svd2rust` output gives an error: 
```
$ rustfmt < stm32f415.rs 2>&1 | less
error: expected `:`, found `}`
 --> stdin:1:31576`
```

Am I correct in thinking the output should have been properly pretty printed (not a giant 7.8M single line)? Is there anything I could look into to figure out why this is happening?

japaric, 2017-04-03:
What svd2rust version were you using? This works for me:

```
$ cargo new --lib stm32f41x && cd $_

$ curl -LO https://raw.githubusercontent.com/posborne/cmsis-svd/master/data/STMicro/STM32F41x.svd

$ svd2rust -i STM32F41x.svd > src/lib.rs

$ rustfmt src/lib.rs

$ cargo add cortex-m

$ cargo add vcell

$ cargo build
```

With `svd2rust 0.5.1`, `cargo 0.19.0-nightly (4e95c6b41 2017-03-23)` and `rustc 1.17.0-nightly (e1cec5d4b 2017-03-29)`.

> Am I correct in thinking the output should have been properly pretty printed (not a giant 7.8M single line)?

No, that's normal (the quasiquoting crate we use under the hood doesn't preserve line breaks and just "inlines" everything). You can run `rustfmt` on the output.

archaelus, 2017-04-03:
I definitely still have problems with `rustfmt 0.8.1 ()` installed (built today with `rustc 1.18.0-nightly (5e122f59b 2017-04-01)`).

I did try patching svdrust to pull in the rustfmt crate and run that before generating output - it's a pretty small change and has generated the `STM32F41x` core properly. I'm guessing I have some weird syntax incompatibility between the svd2rust output and my rustfmt binary.

japaric, 2017-04-06:
I find this very odd. Relatively "contemporary" versions of rustfmt and rustc should parse / understand the same syntax.

I suppose that apart from rustfmt, calling `cargo build` on the unformatted rustfmt output doesn't work either?

I wonder if this because you are passing the code to format through rustfmt stdin (could maybe be some problem with your shell / environment / locale) rather than as passing it to rustfmt as a file.

Do the steps from my previous comment don't work for you either?

japaric, 2017-04-15:
The unformatted output is expected and the [docs](https://docs.rs/svd2rust/0.6.0/svd2rust/) recommend piping svd2rust output before writing it to a file.

And I can't repro the rustfmt / compilation error with the STM32F41x.svd so I'm going to close this. Feel free to open a new issue (with repro steps) if you hit a compiler error with recent versions of svd2rust / rustc.

closed 2017-04-15 by japaric