rust-embedded/svd2rust #69: A Secondary File For Adding Type Information To Fields

cwoodall, 2017-04-03:
Some `svd` files have type information embedded in them with the `enumeratedValues`, but many do not and it is reasonable for the maintainer of an svd2rust memory map to provide these for making a safe and usable interface. Would it be possible to take in a second file of some sort which can associate a Peripheral + Register + Field to a type or inject enumeratedValues? I could see some desire to just say it is safe to write a boolean value to this register, or it is safe to write a u8 to this register without specifying an enum.

The main goal would be to allow for SVDs and Maintainer based meta-data to be maintained separately so that manufacturer provided svd files can be used without editing them directly (which makes it hard to figure out what you added and what was provided for). This also means that as manufacturers update their svd files with more information (if they ever do) the maintainer can more easily update without having to edit the svd file again.

japaric, 2017-04-03:
IMO, we should just inject enumeratedValues into the original SVD file and perhaps change the version stored in the modified file to reflect its non-original status. That way the enhancenments to the SVD files can be reaped not just from Rust but from other programming languages. (Assuming you publish the modified SVD files somewhere public)

> I could see some desire to just say it is safe to write a boolean value to this register, or it is safe to write a u8 to this register without specifying an enum.

I personally prefer enumeratedValues over booleans, the former contain more information about what the bitfield values actually meant.

The number of bitfields that accept the whole `u8` / `u16` / `u32` range is, IMO, too small to warrant creating a new format to add this type information. OTOH, adding more `enumeratedValues` fields to a SVD file gives much more bang for the buck and doesn't require any svd2rust change. 1 / 2 / 3 bits bitfields are very easy to get wrong when `enumeratedValues` info is missing. Whereas, you can't screw up things when the bitfield already accepts the whole u8 range. In this last case, adding type information doesn't prevent misuse (as there isn't any to begin with); it just lets you remove an `unsafe` block that probably it's going to hidden from the end user by some abstraction / driver, in most cases.

> (which makes it hard to figure out what you added and what was provided for).

`diff` or just maintain patches rather than the whole modified file.

cwoodall, 2017-04-10:
I agree with your argument.

Just looking through some of the [SVD documentation](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html#elem_register)  I found some interesting tags, namely:

- `dataType`, seems limited to C data types, and does not really add any guarantees.
- [`writeConstraint`](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html#elem_writeConstraint) is a little more interesting from a safety perspective. It allows registers to be limited only to EnumeratedValues (in rust2svd defaulting to true) or to use ranges. Unfortunately without a range limited integer data type in Rust this adds no real benefit. With one it could be very useful...

Anyway, those are my findings digging a little deeper.

japaric, 2017-04-11:
writeConstraint sounds interesting. I think we can still make it good use of it without range types. Basically, we can use it to assert that a bitfield with N bits can take the whole 0..(1<<N) range and make the write(|w| w.bits(x)) operation safe.

To elaborate, today if we see an SVD like this

``` xml
<register>
  <name>DR</name>
  <fields>
    <field>
      <name>DR</name>
      <bitWidth>16</bitWidth>
..
```

We generate an `unsafe` bits method that takes an `u16`: `dr.write(|w| unsafe { w.dr().bits(1024) })`.

My idea is that if the SVD uses a `writeConstraint` like this:

``` diff
       <bitWidth>16</bitWidth>
+      <writeConstraint>
+        <range>
+          <minimum>0</minimum>
+          <maximum>65535</maximum>
+        </range>
+      </writeConstraint>
 ..
```

then we can make the bits method safe: `dr.write(|w| w.dr.bits(1024))`

We can apply this idea to any field regardless of its `bitWidth` value as we internally mask the bits input to fit into the bitfield. For instance, if the above DR had a bitWidth of 9 then an operation like `dr.write(|w| w.dr.bits(513))` would still be safe (it would write a `1` to the bitfield) as it's writing a *valid* value to the bitfield. (It may be a logic error, so a warning / compiler error would be ideal but that requires proper range types)

cwoodall, 2017-04-11:
I like it. For me the main benefit is that the thing that is closest to the datasheet is carrying over as much information as possible (even if some of it is user provided). I would be willing to take a stab at this myself, but I might not have much time to work on it until next Tuesday.

cwoodall, 2017-04-22:
Started  by adding support to https://github.com/japaric/svd

japaric, 2017-04-25:
@cwoodall do you consider this fixed by #73?

cwoodall, 2017-04-25:
Yep!

japaric, 2017-04-25:
Alright, let's close this then.

closed 2017-04-25 by japaric