rust-embedded/svd2rust #75: Reset value should not be required

whitequark, 2017-04-29:
The SVD spec does not require reset values to be present. This happens in real data, e.g. on TM4C series it looks like all writable registers are always initialized to 0.

whitequark, 2017-04-29:
Actually a parser bug.

whitequark, 2017-04-29:
Sorry, I confused myself somehow between projects, it is not a parser bug. Here is the workaround:
```rust
diff --git a/src/generate.rs b/src/generate.rs
index 06f9d13..bb1593b 100644
--- a/src/generate.rs
+++ b/src/generate.rs
@@ -494,11 +496,8 @@ pub fn register(
 
         let rv = register.reset_value
             .or(defs.reset_value)
-            .map(|rv| util::unsuffixed(u64(rv)))
-            .ok_or_else(|| {
-                            format!("Register {} has no reset value",
-                                    register.name)
-                        })?;
+            .or(Some(0))
+            .map(|rv| util::unsuffixed(u64(rv)));
 
         w_impl_items.push(
             quote! {
```

zargony, 2017-04-29:
Same problem happened to me when trying to parse the Atmel SAM3 SVD files. They don't seem to contain reset values for write-only registers.

japaric, 2017-04-29:
> Here is the workaround:

Seems OK to me, could you send a PR, @whitequark?

zargony, 2017-04-29:
While this patch works, it also means that all registers will get a `reset_value()` method that returns 0, which may be confusing since the reset value is actually unspecified.

Maybe it would be more accurate to not generate a `reset_value()` method at all in this cases. I'm new to svd2rust, so I'm not sure how big of a change this would be. As far as I see, some other methods depend on `reset_value`.

zargony, 2017-04-29:
Browsed the source again and it seems that `write` is the only other method that depends on a reset value.

whitequark, 2017-04-29:
> While this patch works, it also means that all registers will get a reset_value() method that returns 0, which may be confusing since the reset value is actually unspecified.

I'm not quite sure how to handle this case. In principle, we could proceed as follows:
  * If the reset value of all (writable) bits is 0, like on TM4C, then that chip should specify that at `<device>` level.
  * If the reset value is, however, *unspecified*, then only RMW cycles are available, thus more strictly following the manufacturer's recommendation that all unused bits are written the same value that is read from them.

The questions to answer here would be:
  1. What is the semantics SVD specifies in absence of `<resetValue>` at any level? I could not find it in the ARM documentation.
  2. Even in the latter case, we have to allow pure write cycles to write-only registers, so that'll have to be special-cased anyway.
  3. How important it is in practical reality to religiously do RMW cycles? I know it's strongly recommended, and I know there are occasional bugs stemming from them, but it's very anecdotal. Do vendors *really* still add bits in registers that are 1 at reset and have to stay that way, or add bits that RW0C? That seems like a really awful decision, and while I believe that *some* vendor somewhere *is* going to be that stupid, should we build the entire svd2rust defending from them?

whitequark, 2017-04-29:
I thought about it and I think that building some complicated procedure around an easily worked around potential problem is not worth the time spent. So, I will send a PR with the patch above.

zargony, 2017-04-29:
I agree that without a reset value we theoretically had to enforce RMW cycles, however that'd be very inconvenient.
But how about only providing a `reset_value()` method if a reset value is specified? This would error at compile time if somebody tries to get a reset value that doesn't exist and would be a cleaner API than falsely returning zero. Writing a value should then start with the reset value if specified or with 0 otherwise for convenience.

whitequark, 2017-04-29:
> But how about only providing a reset_value() method if a reset value is specified? This would error at compile time if somebody tries to get a reset value that doesn't exist and would be a cleaner [...]

This seems odd as the current idiomatic way to set a register to its reset value is to use `.write(|w| w)` and that's what the `.reset()` shorthand is going to use (#19)...

zargony, 2017-04-29:
Sorry, I'm quite new to this project and don't know what's idiomatic very well yet.

A `reset()` method as a shorthand sounds good, but would also only make sense if a reset value exists. I'd propose to also not provide `reset()` if no reset value is present.

whitequark, 2017-04-29:
I'll leave this decision to @japaric.

Emilgardis, 2017-04-29:
I agree with @zargony. If the svd provides a resetValue, then we know we can "trust"

whitequark, 2017-04-29:
> If the svd provides a resetValue, then we know we can "trust" it.

But that's not the issue at hand. The issue is what to do if it *doesn't*.

whitequark, 2017-04-29:
As mentioned by @Emilgardis in #79 the reset value *is* in fact required (although the SVD schema does not mandate that, probably because XSD is not powerful enough).

closed 2017-04-29 by whitequark