rust-embedded/svd2rust #76: Name mangling is lossy around numbers

whitequark, 2017-04-29:
E.g. both `ADC_SPC_PHASE_22_5` and `ADC_SPC_PHASE_225` converge to `AdcSpcPhase225`.

whitequark, 2017-04-29:
I use the following patch to the `inflections` crate as a workaround:
```patch
diff --git a/src/case.rs b/src/case.rs
index 0c6bea7..a22677f 100644
--- a/src/case.rs
+++ b/src/case.rs
@@ -443,6 +443,12 @@ pub fn is_constant_case(string: &str) -> bool {
   string == to_constant_case(string)
 }
 
+/// Checks if a character is a digit.
+#[inline]
+fn is_digit(c: char) -> bool {
+  c >= '0' && c <= '9'
+}
+
 /// Checks if a character is a separator.
 #[inline]
 fn is_separator(c: char) -> bool {
@@ -472,7 +478,15 @@ fn scan_to_camel(state: &mut (bool, Option<char>), curr: char) -> Option<String>
   let last = state.1;
   state.1 = Some(curr);
 
-  if state.0 {
+  if state.0 && is_digit(curr) {
+    // If the state has signaled the next character must be capitalized,
+    // but it is a digit, then prepend a separator to it.
+    state.0 = false;
+    let mut string = String::with_capacity(2);
+    string.push('_');
+    string.push(curr);
+    Some(string)
+  } else if state.0 {
     // If the state has signaled the next character must be capitalized,
     // capitalize it and mark the state as finished.
     state.0 = false;
```

whitequark, 2017-04-29:
Note: this is not perfect and results in errors such as follows:
```
error: variant `CcmCrcctrlInit_0` should have a camel case name such as `Ccmcrcctrlinit0`
```

whitequark, 2017-04-29:
The more I think about it, the more I realize that name mangling of types is a bad idea.
  1. It creates complications like this issue.
  2. It is literally impossible to satisfy the `non_camel_case_types` lint without destroying semantics sometimes, because names like `Foo4_20` are invalid:
    ```
    warning: type `Foo4_20` should have a camel case name such as `Foo420`
    ```
  3. The specific naming of the type matters rarely because it never appears in user's code.
  4. When the type name *does* matter is when it appears in the error messages, and I cannot in good faith describe names like `SysctlSri2cR0` as in any way easy to read. It simply goes against any established practice in embedded dev.

Therefore I propose to allow non-camel-case types and stop this mangling altogether.

Emilgardis, 2017-04-30:
All \<name\> tags are supposed to follow C ansi code standards. Code gen from SVD's usually use macro naming conventions, i.e exactly what \<name\> holds. This issue is a tricky one to fix.

Can one possibly to make `22_5` into something that conveys the implicit `.`?

whitequark, 2017-04-30:
> Can one possibly make 22_5 into something that conveys the implicit .?

One could of course mangle the SVD files in any applicable way to work around a deficiency of svd2rust but I don't think this should be necessary.

commit ae89a86d (2017-06-05): Auto merge of #85 - whitequark:uppercase, r=japaric

Normalize SVD names to upper case where they were to pascal case

Fixes #76.

Depends on https://github.com/calebmer/inflections/pull/2.
commit d3fb2813 (2017-06-05): Normalize SVD names to upper case where they were to pascal case.

Fixes #76.
PR #85 (2017-06-05): Normalize SVD names to upper case where they were to pascal case
Fixes #76.

Depends on https://github.com/calebmer/inflections/pull/2.
closed 2017-06-05 by homunkulus