rust-embedded/svd2rust #87: Registers with multiple reserved fields causes duplicate definitions

simonvandel, 2017-05-09:
Running the tool on [this svd file](https://gist.github.com/simonvandel/6a67b7893f89482b8be2b7ac75a661b6) generates code that does not compile.

The error message reads:
```
error[E0201]: duplicate definitions with name `reserved`:
      --> src/lib.rs:156931:13
       |
156881 | /             pub fn reserved(&self) -> ReservedR {
156882 | |                 let bits = {
156883 | |                     const MASK: u8 = 1;
156884 | |                     const OFFSET: u8 = 0;
...      |
156887 | |                 ReservedR { bits }
156888 | |             }
       | |_____________- previous definition of `reserved` here
...
156931 | /             pub fn reserved(&self) -> ReservedR {
156932 | |                 let bits = {
156933 | |                     const MASK: u32 = 134217727;
156934 | |                     const OFFSET: u8 = 5;
...      |
156937 | |                 ReservedR { bits }
156938 | |             }
       | |_____________^ duplicate definition

```

This happens as a lot of registers in the svd file has multiple reserved fields.

I see two different solutions to this problem:
1. Edit the SVD file, appending a unique string to each conflicting RESERVED field.
2. Change svd2rust, by special casing registers  with multiple reserved fields such that unique field names are generated. i.e. RESERVED1, RESERVED2

What do you think?

pftbest, 2017-05-09:
This doesn't look right, reserved fields should not be present in svd file.

japaric, 2017-05-09:
Reserved bitfields shouldn't be exposed in the API. As they are reserved they shouldn't be modifiable.

I ... thought we already did something like this. But maybe that was for registers named reserved and not for bitfields.

pftbest, 2017-05-09:
No, I'm wrong. The svd spec allows it:

> The case-insensitive field named "reserved" is treated as a keyword and each field with this name is ignored

simonvandel, 2017-05-09:
The SVD is downloaded directly from [here](https://github.com/posborne/cmsis-svd/blob/master/data/NXP/LPC176x5x_v0.2.svd) with the following patch applied.

```patch
203c203
< 									
---
> 									<value>0x3</value>
```

simonvandel, 2017-05-10:
The patch sets a value for an enumerated value that is reserved, as svd2rust otherwise refuses to generate code. This also seems like a bug. I will open a separate issue.

commit 2c34e261 (2017-05-16): Avoid generating code for reserved fields and enumerated values. Fixes #87
PR #89 (2017-05-16): Avoid generating code for reserved fields and enumerated values.
This is my attempt to fix #87.

Code is still being generated for reserved, but this reduces most cases.
closed 2017-05-16 by homunkulus