rust-embedded/svd2rust #91: support <access>read-writeOnce</access>

japaric, 2017-05-22:
The code generated from the file [STM32L063x.svd](https://github.com/posborne/cmsis-svd/blob/python-0.4/data/STMicro/STM32L063x.svd) doesn't compile. Because the read-writeOnce access mode is not supported by svd2rust the DAC_SR register has no access methods:

``` rust
# [ doc = "Digital-to-analog converter" ]
pub mod dac {
    // ..
    # [ doc = "status register" ]
    pub struct Sr {
        register: VolatileCell<u32>,
    }
    # [ doc = "status register" ]
    pub mod sr {
        impl super::Sr {}
    }
}
```

And you run into an unused field warning which becomes an error because of `#![deny(warnings)]`.

kjetilkjeka, 2017-05-23:
I will contribute to this issue instead of making a new since I suspect these issues are the same.

When using svd2rust on NXP S32K144 svd file i get:

    svd2rust -i ~/Nedlastinger/S32K144.svd | rustfmt | tee src/s32k144.rs
    thread 'main' panicked at 'unknown access variant: writeonce', 
    /home/kjeka/.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.5.1/src/lib.rs:286
    note: Run with `RUST_BACKTRACE=1` for a backtrace.

and with RUST_BACKTRACE=1

    RUST_BACKTRACE=1 svd2rust -i ~/Nedlastinger/S32K144.svd | rustfmt | tee src/s32k144.rs
    thread 'main' panicked at 'unknown access variant: writeonce',         
    /home/kjeka/.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.5.1/src/lib.rs:286
    stack backtrace:
       0: std::sys::imp::backtrace::tracing::imp::unwind_backtrace
                 at /checkout/src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
       1: std::sys_common::backtrace::_print
                 at /checkout/src/libstd/sys_common/backtrace.rs:71
       2: std::panicking::default_hook::{{closure}}
                 at /checkout/src/libstd/sys_common/backtrace.rs:60
                 at /checkout/src/libstd/panicking.rs:355
       3: std::panicking::default_hook
                 at /checkout/src/libstd/panicking.rs:371
       4: std::panicking::rust_panic_with_hook
                 at /checkout/src/libstd/panicking.rs:549
       5: std::panicking::begin_panic
                 at /checkout/src/libstd/panicking.rs:511
       6: std::panicking::begin_panic_fmt
                 at /checkout/src/libstd/panicking.rs:495
       7: <core::option::Option<T>>::map
       8: <core::iter::FilterMap<I, F> as core::iter::iterator::Iterator>::next
       9: svd_parser::Device::parse
      10: svd_parser::parse
      11: svd2rust::main
      12: __rust_maybe_catch_panic
                 at /checkout/src/libpanic_unwind/lib.rs:98
      13: std::rt::lang_start
                 at /checkout/src/libstd/panicking.rs:433
                 at /checkout/src/libstd/panic.rs:361
                 at /checkout/src/libstd/rt.rs:56
      14: __libc_start_main
      15: _start

I will attach the file in case you would like to test for yourself.
[S32K144.svd.zip](https://github.com/japaric/svd2rust/files/1022193/S32K144.svd.zip)

Emilgardis, 2017-05-23:
@kjetilkjeka Your error is not related, it comes from a miswritten svd file, the variant is called `writeOnce`, not `writeonce`. We could fix this in svd-parser, but I think we should strive to follow the SVD 1.0/1.1 schema as close as possible.

To fix your error and generate some code simply patch your svd to change all the `writeonce` into `writeOnce`.

kjetilkjeka, 2017-05-24:
@Emilgardis You are of course right. It worked when replacing writeonce with writeOnce, thanks.

And for what it's worth I do agree with your view on forcing the svd to be up to spec rather than accepting badly written svd files.

adamgreig, 2019-07-26:
Implemented by #322.

closed 2019-07-26 by adamgreig