rust-embedded/svd2rust #92: "bits" is a valid field name

japaric, 2017-05-22:
So the method to access the bitfield proxy:

``` rust
peripheral.register.read().bits() // BitsR
peripheral.register.read().bits().is_variant() // bool
```

becomes ambiguous with the register proxy method to get its raw bits

``` rust
peripheral.register.read().bits()  // e.g. u8
```

This happens in the [SKEAZN84.svd](https://github.com/posborne/cmsis-svd/blob/python-0.4/data/Freescale/SKEAZN84.svd) file.

japaric, 2017-05-25:
### Solution

Rename `R.bits` to something else. The question is what should be the new name. In principle any name we pick can cause a collision in the future as register names, per the SVD specification, are as, or more, free-from than Rust identifiers.

From the schema (register names have type `dimableIdentifierType`):

``` xml
  <xs:simpleType name="dimableIdentifierType">
    <xs:restriction base="xs:string">
      <xs:pattern value="((%s)|(%s)[_A-Za-z]{1}[_A-Za-z0-9]*)|([_A-Za-z]{1}[_A-Za-z0-9]*(\[%s\])?)|([_A-Za-z]{1}[_A-Za-z0-9]*(%s)?[_A-Za-z0-9]*)"/>
    </xs:restriction>
  </xs:simpleType>
```

zargony, 2017-05-26:
Also happened to me with [ATSAM3X8E.svd](https://github.com/posborne/cmsis-svd/blob/76d64889ab00c5290773e639265e147d7b3458ea/data/Atmel/ATSAM3X8E.svd). My workaround was to rename the field rather than the method (which works, but also feels wrong)

japaric, 2017-06-06:
We can use [@whitequark's idea]](https://github.com/japaric/svd2rust/pull/99#issuecomment-306233776) of renaming the 'bits` variant (when the enumeratedValue is named BITS) iff it would cause a collision with R.bits.

commit 9ed581c9 (2017-06-06): add regression tests for #92 and #99

closes #92
PR #110 (2017-06-06): add regression tests for #92 and #99
closes #92
closed 2017-06-06 by homunkulus