rust-embedded/svd2rust #102: register of size 1?

japaric, 2017-06-05:
File: MB9AF10xN.svd

Contents:

``` xml
        <register>
          <name>WDG_RIS</name>
          <description>Hardware Watchdog Timer Interrupt Status Register</description>
          <addressOffset>0x10</addressOffset>
          <size>1</size>
          <access>read-only</access>
          <resetValue>0xFF</resetValue>
          <resetMask>0x00</resetMask>
          <!-- FIELDS -->
          <fields>
            <!-- FIELD "RIS" -->
            <field>
              <name>RIS</name>
              <description>Hardware watchdog interrupt status bit</description>
              <lsb>0</lsb>
              <msb>0</msb>
              <access>read-only</access>
            </field>
          </fields>
        </register>
```

Produces:

``` rust
        impl R {
            # [ doc = r" Value of the register as raw bits" ]
            # [ inline ( always ) ]
            pub fn bits(&self) -> bool {
                self.bits
            }
            # [ doc = "Bit 0 - Hardware watchdog interrupt status bit" ]
            # [ inline ( always ) ]
            pub fn ris(&self) -> RISR {
                let bits = {
                    const MASK: bool = true;
                    const OFFSET: u8 = 0;
                    ((self.bits >> OFFSET) & MASK as bool) != 0
                };
                RISR { bits }
            }
        }
```

Compiler error:

``` console
error[E0369]: binary operation `>>` cannot be applied to type `bool`
    --> src/lib.rs:6496:22
     |
6496 |                     ((self.bits >> OFFSET) & MASK as bool) != 0
     |                      ^^^^^^^^^^^^^^^^^^^^^
     |
     = note: an implementation of `std::ops::Shr` might be missing for `bool`

error: aborting due to previous error(s)
```

The problem I see that the SVD file declares the register has having a size of 1. I'm not sure if that allowed by the SVD format (the website appears to be down ATM) but I'd expect it to only allow sizes multiple of 8 (bytes). Note that we are talking about *registers*; bitfields can have any size (smaller than the size of their register).

Anyhow the current implementation maps a size of 1 to a `bool` so the `bits` field of `R` is `bool` which can be shifted to the right by `OFFSET` thus the compiler error.

cc @whitequark fallout from #84

whitequark, 2017-06-05:
> I'm not sure if that allowed by the SVD format (the website appears to be down ATM)

I can't find anything in the SVD spec that disallows this, but it doesn't make sense to me to accept it unless the CPU can perform 1-bit reads and writes. Well, certain CPUs can; e.g. Atmel AVR can do sbi/cbi, and you could do bit-banding on Cortex-M3+, but it's weird.

japaric, 2017-06-05:
> but it doesn't make sense to me to accept it 

I agree.

I think we should round up the register size to be a multiple of 8 (bits). That should fix this problem.

whitequark, 2017-06-05:
To a power of two multiple of 8, perhaps?

japaric, 2017-06-05:
Yeah, that's what I was thinking but didn't fully convey in my previous comment.

commit 4c890041 (2017-06-06): make sure that the register size is always a power of 2 >= 8

fixes #102
PR #106 (2017-06-06): make sure that the register size is always a power of 2 >= 8
fixes #102

cc @whitequark
waiting for #104 to make sure this doesn't break other stuff
closed 2017-06-06 by homunkulus