rust-embedded/svd2rust #105: Fields `Not_Used` and `NOT_USED` map to the same type

japaric, 2017-06-05:
File: MK61F15.svd

Contents:

``` xml
          <fields>
            <field>
              <name>Not_Used</name>
              <description>Reserved</description>
              <bitOffset>0</bitOffset>
              <bitWidth>16</bitWidth>
              <access>read-only</access>
            </field>
            <field>
              <name>NOT_USED</name>
              <description>Reserved</description>
              <bitOffset>16</bitOffset>
              <bitWidth>16</bitWidth>
              <access>read-only</access>
            </field>
          </fields>
```

Error:

``` console
error[E0428]: a type named `NOT_USEDR` has already been defined in this module
      --> src/lib.rs:530278:9
       |
530267 | /         pub struct NOT_USEDR {
530268 | |             bits: u16,
530269 | |         }
       | |_________- previous definition of `NOT_USEDR` here
...
530278 | /         pub struct NOT_USEDR {
530279 | |             bits: u16,
530280 | |         }
       | |_________^ `NOT_USEDR` already defined
```

The read proxy for these two fields, `Not_Used` and `NOT_USED`, is `NOT_USEDR`. I don't know if this is something we want to support. This SVD file is also kind wrong because it should be using the keyword RESERVED for reserved bitfields then svd2rust will simply had ignored the bitfields.

Should we support this or flag the SVD file as being broken?

cc @whitequark

whitequark, 2017-06-05:
Seems clearly broken to me; while the spec does not explicitly say that uniqueness should be modulo case, I believe that the C code generators for CMSIS assume that and casefold, and this file only works because of how loose the C standard is on this point.

japaric, 2017-06-06:
Thank for the input @whitequark.

Closing as "this is not a bug in svd2rust but in the SVD files".

closed 2017-06-06 by japaric