rust-embedded/svd2rust #107: support clusters

japaric, 2017-06-05:
Example:

File: ATSAMD21E15A.svd

Contents:

``` xml
    <peripheral>
      <name>SERCOM0</name>
      <version>2.0.0</version>
      <description>Serial Communication Interface 0</description>
      <groupName>SERCOM</groupName>
      <prependToName>SERCOM_</prependToName>
      <baseAddress>0x42000800</baseAddress>
      <addressBlock>
        <offset>0</offset>
        <size>0x40</size>
        <usage>registers</usage>
      </addressBlock>
      <interrupt>
        <name>SERCOM0</name>
        <value>9</value>
      </interrupt>
      <registers>
       <cluster>
        <name>I2CM</name>
        <description>I2C Master Mode</description>
        <headerStructName>SercomI2cm</headerStructName>
        <addressOffset>0x0</addressOffset>
        <register>
          <name>ADDR</name>
          <!-- .. -->
       </cluster> <!-- SercomI2cm -->
       <cluster>
        <name>I2CS</name>
        <description>I2C Slave Mode</description>
        <alternateCluster>I2CM</alternateCluster>
        <headerStructName>SercomI2cs</headerStructName>
        <addressOffset>0x0</addressOffset>
        <register>
          <name>ADDR</name>
          <!-- .. -->
```

These clusters seem to be non-functional. They only add more structure (namespaces) to a register block. The above is supposed to generate something like this:

``` rust
pub struct RegisterBlock {
    pub i2cm: I2CM,
    pub i2cs: I2CS,
    ..
}

pub struct I2CM {
    pub addr: ADDR,
    ..
}

pub struct I2CS {
    pub addr: ADDR,
    ..
}
```

The "flat" version would have resulted in a name collision between I2CM.ADDR and I2CS.ADDR.

We currently don't generate anything at all for clusters. For instance the above SERCOM0 peripheral is totally omitted in the output because it appears to have 0 registers as clusters are ignored.

japaric, 2017-06-06:
Blocker: svd doesn't parse `<cluster>` elements. See japaric/svd#33

therealprof, 2017-09-28:
Is anyone working on this at the moment?

jamesmunns, 2018-02-24:
This also affects https://github.com/jamesmunns/nrf52. I may look into implementing clusters.

jamesmunns, 2018-02-27:
@japaric - how should `svd2rust` handle clusters that contain registers with repeated names in the same scope? Particularly ones like:

```xml
       <cluster>
        <name>USART</name>
        <description>USART Mode</description>
        <alternateCluster>I2CM</alternateCluster>
        <headerStructName>SercomUsart</headerStructName>
        <addressOffset>0x0</addressOffset>
        <register>
          <name>BAUD</name>
          <description>USART Baud Rate</description>
          <alternateGroup>DEFAULT_MODE</alternateGroup>
          <addressOffset>0x0C</addressOffset>
          <size>16</size>
          <fields>
            <field>
              <name>BAUD</name>
              <description>Baud Rate Value</description>
              <bitOffset>0</bitOffset>
              <bitWidth>16</bitWidth>
            </field>
          </fields>
        </register>
        <register>
          <name>BAUD</name>
          <description>USART Baud Rate</description>
          <alternateGroup>FRAC_MODE</alternateGroup>
          <addressOffset>0x0C</addressOffset>
          <size>16</size>
          <fields>
            <field>
              <name>BAUD</name>
              <description>Baud Rate Value</description>
              <bitOffset>0</bitOffset>
              <bitWidth>13</bitWidth>
            </field>
            <field>
              <name>FP</name>
              <description>Fractional Part</description>
              <bitOffset>13</bitOffset>
              <bitWidth>3</bitWidth>
            </field>
          </fields>
        </register>
        <register>
```
Is there a tracking issue for `union`'d registers (inside or not inside clusters) in `svd2rust`? is this the same as https://github.com/japaric/svd2rust/issues/16 ?

closed 2018-03-09 by jamesmunns