rust-embedded/svd2rust #112: Fail to compile for LPC176x5x_v0.2.svd

nicholastmosher, 2017-06-08:
Hi, I'm trying to create a crate for the [LPC1769](https://developer.mbed.org/platforms/mbed-LPC1768/) using the [LPC176x5x_v0.2.svd](https://github.com/posborne/cmsis-svd/blob/master/data/NXP/LPC176x5x_v0.2.svd) file. The output `lib` file seems to have a lot of errors involving duplicate definitions. Running `svd2rust` as in the [usage](https://docs.rs/svd2rust/0.9.1/svd2rust/#usage) works just fine, but when I try to compile the crate, I get the following errors:

```
$ cargo build
   Compiling LPC1769 v0.1.0 (file:///home/nick/Documents/LPC1769)
error[E0428]: a type named `MR` has already been defined in this module
     --> src/lib.rs:22536:5
      |
20407 | /     pub struct MR {
20408 | |         register: VolatileCell<u32>,
20409 | |     }
      | |_____- previous definition of `MR` here
...
22536 | /     pub struct MR {
22537 | |         register: VolatileCell<u32>,
22538 | |     }
      | |_____^ `MR` already defined

error[E0428]: a module named `mr` has already been defined in this module
     --> src/lib.rs:22540:5
      |
20411 | /     pub mod mr {
20412 | |         # [ doc = r" Value read from the register" ]
20413 | |         pub struct R {
20414 | |             bits: u32,
...     |
20512 | |         }
20513 | |     }
      | |_____- previous definition of `mr` here
...
22540 | /     pub mod mr {
22541 | |         # [ doc = r" Value read from the register" ]
22542 | |         pub struct R {
22543 | |             bits: u32,
...     |
22641 | |         }
22642 | |     }
      | |_____^ `mr` already defined

error[E0428]: a type named `THE_TIME_COUNTERS_AR` has already been defined in this enum
     --> src/lib.rs:28515:13
      |
28513 |             THE_TIME_COUNTERS_AR,
      |             -------------------- previous definition of `THE_TIME_COUNTERS_AR` here
28514 |             # [ doc = "The time counters are disabled so that they may be initialized." ]
28515 |             THE_TIME_COUNTERS_AR,
      |             ^^^^^^^^^^^^^^^^^^^^ `THE_TIME_COUNTERS_AR` already defined

error[E0428]: a type named `THE_CALIBRATION_COUN` has already been defined in this enum
     --> src/lib.rs:28615:13
      |
28613 |             THE_CALIBRATION_COUN,
      |             -------------------- previous definition of `THE_CALIBRATION_COUN` here
28614 |             # [ doc = "The calibration counter is enabled and counting, using the 1 Hz clock. When the calibration counter is equal to the value of the CALIBRATION register, the counter resets and repeats counting up to the value of the CALIBRATION register. See Section 30.6.4.2 and  Section 30.6.5." ]
28615 |             THE_CALIBRATION_COUN,
      |             ^^^^^^^^^^^^^^^^^^^^ `THE_CALIBRATION_COUN` already defined

error[E0428]: a type named `DISABLED` has already been defined in this enum
     --> src/lib.rs:52911:13
      |
52909 |             DISABLED,
      |             -------- previous definition of `DISABLED` here
52910 |             # [ doc = "Disabled. P0.6 pin has neither pull-up nor pull-down." ]
52911 |             DISABLED,
      |             ^^^^^^^^ `DISABLED` already defined

error[E0428]: a type named `CONFIG` has already been defined in this module
      --> src/lib.rs:135025:5
       |
132879 | /     pub struct CONFIG {
[... 316 lines cut ...]
      | |_____________^ duplicate definition

error[E0201]: duplicate definitions with name `is_error_occurred_durin`:
     --> src/lib.rs:80433:13
      |
80428 | /             pub fn is_error_occurred_durin(&self) -> bool {
80429 | |                 *self == ERRDIRR::ERROR_OCCURRED_DURIN
80430 | |             }
      | |_____________- previous definition of `is_error_occurred_durin` here
...
80433 | /             pub fn is_error_occurred_durin(&self) -> bool {
80434 | |                 *self == ERRDIRR::ERROR_OCCURRED_DURIN
80435 | |             }
      | |_____________^ duplicate definition

error: aborting due to previous error(s)

error: Could not compile `LPC1769`.

To learn more, run the command again with --verbose.
```

I apologize if this isn't the right place to report this. Do you have any suggestions on how to fix this*, or do you know of any existing support crate for the LPC1769? Any help would be greatly appreciated!

*I wondered about simply deleting the duplicate code, but I'm worried that may cause other problems that might be harder to detect, plus that would be more of a band-aid solution.

roysmeding, 2017-06-08:
Hi!
I looked through the SVD file, and the errors about `struct mr` look like the same issue described in #90.

(Most of?) the other errors seem due to errors in the input SVD file. It looks like they generated the enum names using a tool that cuts off long names, which causes some of them to be identical.

e.g. on line 5476-5488 of the SVD file:

```
					<enumeratedValues>
					<name>ENUM</name>
						<enumeratedValue>
							<name>THE_TIME_COUNTERS_AR</name>
							<description>The time counters are enabled.</description>
							<value>1</value>
						</enumeratedValue>
						<enumeratedValue>
							<name>THE_TIME_COUNTERS_AR</name>
							<description>The time counters are disabled so that they may be initialized.</description>
							<value>0</value>
						</enumeratedValue>
					</enumeratedValues>
```

These issues can only be fixed in the SVD file (or by finding a file for your device that doesn't have these issues.)

It would be good if svd2rust checked for this, though, instead of passing the names through verbatim.

nicholastmosher, 2017-06-08:
Ah, thanks for taking the time to answer, that helps a lot! Do you know if there's a workaround for the MR problem at the moment? I imagine I can at least complete all of the enum names to fix that part.

I'll go ahead and close this since all of the problems here seem to be represented by other issues or are specific to my scenario and not svd2rust :)

closed 2017-06-08 by nicholastmosher

roysmeding, 2017-06-08:
AIUI, it should be possible to bypass the issue with non-contiguous arrays by writing a separate definition for each register instead of using the SVD array functionality.

In other words, instead of the two `<register>` definitions for `MR%s` that use `<dim>` and `<dimIndex>` to generate multiple registers, you'd have one `<register>` for each individual one, named e.g. MR0, MR1, etc. (it seems there are six of them total).

nicholastmosher, 2017-06-08:
That worked, thanks a bunch! Now the only thing I have to do is figure out what's causing those duplicate function definitions and I should be set!

nicholastmosher, 2017-06-08:
For posterity, the way I got rid of the errors with duplicate `is_set` functions is by removing the `enumeratedValues` from the fields in the registers with the clashes. For example, in the ICR register, each field looked like:

```
<field>
    <name>TI3</name>
    <description>Transmit Interrupt 3. This bit is set when the TBS3 bit in CANxSR goes from 0 to 1 (whenever a message out of TXB3 was successfully transmitted or aborted), indicating that Transmit buffer 3 is available, and the TIE3 bit in CANxIER is 1.</description>
    <bitRange>[10:10]</bitRange>
    <enumeratedValues>
        <name>ENUM</name>
        <enumeratedValue>
            <name>RESET</name>
            <description>Reset</description>
            <value>0</value>
        </enumeratedValue>
        <enumeratedValue>
            <name>SET</name>
            <description>Set</description>
            <value>1</value>
        </enumeratedValue>
    </enumeratedValues>
</field>
```

Since one of those enumerations was named exactly `SET`, svd2rust would create an accessor function named after it, which turned into `is_set`, which was already a function that svd2rust placed in the interface, causing the clash. After fixing the entries, the field looks like this:

```
<field>
    <name>TI3</name>
    <description>Transmit Interrupt 3. This bit is set when the TBS3 bit in CANxSR goes from 0 to 1 (whenever a message out of TXB3 was successfully transmitted or aborted), indicating that Transmit buffer 3 is available, and the TIE3 bit in CANxIER is 1.</description>
    <bitRange>[10:10]</bitRange>
</field>
```

japaric, 2017-06-10:
>  the way I got rid of the errors with duplicate is_set functions 

FWIW, that has already been fixed in #109 but has not made it into a release yet. I'm waiting on making a decision on #111 before I make a release. If you could comment on that issue I'd appreciate it.

simonvandel, 2017-06-11:
@nicholastmosher can you share the SVD file you got working?

nicholastmosher, 2017-06-11:
@simonvandel of course! I probably should have done that in my last post. I've gone ahead and made a repository with the svd2rust generated crate, it has the SVD file in the root directory: https://github.com/nicholastmosher/lpc176x5x-rs. The only other change I've made to the crate is that I broke up the top-level modules into their own files in order to help intellij and the intellij-rust plugin do the smart-complete magic :)

simonvandel, 2017-06-12:
@nicholastmosher Thanks!