rust-embedded/svd2rust #114: Merge registers with the same type into an array

japaric, 2017-06-10:
The code generated for the GPIO peripheral of nrf51 micros looks like this:

``` rust
pub struct Gpio {
    pub out: Out,
    pub outset: Outset,
    pub outclr: Outclr,
    pub in_: In,
    pub dir: Dir,
    pub dirset: Dirset,
    pub dirclr: Dirclr,
    pub pin_cnf0: PinCnf,
    pub pin_cnf1: PinCnf,
    pub pin_cnf2: PinCnf,
    pub pin_cnf3: PinCnf,
    pub pin_cnf4: PinCnf,
    pub pin_cnf5: PinCnf,
    pub pin_cnf6: PinCnf,
    pub pin_cnf7: PinCnf,
    pub pin_cnf8: PinCnf,
    pub pin_cnf9: PinCnf,
    pub pin_cnf10: PinCnf,
    pub pin_cnf11: PinCnf,
    pub pin_cnf12: PinCnf,
    pub pin_cnf13: PinCnf,
    pub pin_cnf14: PinCnf,
    pub pin_cnf15: PinCnf,
    pub pin_cnf16: PinCnf,
    pub pin_cnf17: PinCnf,
    pub pin_cnf18: PinCnf,
    pub pin_cnf19: PinCnf,
    pub pin_cnf20: PinCnf,
    pub pin_cnf21: PinCnf,
    pub pin_cnf22: PinCnf,
    pub pin_cnf23: PinCnf,
    pub pin_cnf24: PinCnf,
    pub pin_cnf25: PinCnf,
    pub pin_cnf26: PinCnf,
    pub pin_cnf27: PinCnf,
    pub pin_cnf28: PinCnf,
    pub pin_cnf29: PinCnf,
    pub pin_cnf30: PinCnf,
    pub pin_cnf31: PinCnf,
    // some fields omitted
}
```

Instead, it should generate:

``` rust
pub struct Gpio {
    pub out: Out,
    pub outset: Outset,
    pub outclr: Outclr,
    pub in_: In,
    pub dir: Dir,
    pub dirset: Dirset,
    pub dirclr: Dirclr,
    pub pin_cnf: [PinCnf; 32],
    // some fields omitted
}
```
The SVD file contains information about PinCnf being an array but the current svd2rust implementation simply *unrolls* the array into several struct fields instead of creating a single struct field that represents the array of registers.

kjetilkjeka, 2017-07-04:
When you write "contains information about PinCnf beeing an array" I assume you're talking about the dim options? Or are there other cases where this grouping should be done?

kjetilkjeka, 2017-07-04:
Ignore my previous comment, I wrote it before I looked into the project and how arrays are expanded.

Regarding the issue to have the RegisterBlock using arrays instead on unrolling. There are at least two ways to achieve this.

1. Instead of expanding the array, make it into a rust array. Some problems regarding this solution
    - What do we do about svd arrays not starting from 0, or skipping elements (```<dimIndex> 3-6 </dimIndex>```)
   - What do we do about literal indexes? (```<dimIndex> A, B, C, D </dimIndex```>)

2. After expanding unroll the fields into an array if they adhere to some criteria (same type, same name except ending, numbers start from 0, no holes, etc)

I guess no 2. is the way you're suggesting but isn't it better to simply not unroll when some criterions are satisfied instead of reversing the work already done?

The way that makes the most sense to me is to make an array if there are integral indexes starting from 0 and contains no holes?

Anyway, I'm in need of this feature and willing to implement it. If you think that is ok, I will need a "go on with 1" or "2 is better because of X" and I can create a PR and start the implementation.

kjetilkjeka, 2017-07-04:
One more remark:

Method 2 won't unroll registers if they're not grouped in SVD as they will create different register types. For an example, you can look at, https://docs.rs/s32k144/0.2.0/s32k144/can0/struct.RegisterBlock.html

So Method 1 and 2 will most likely be identical for the user.

PR #121 (2017-09-20): Register array in RegisterBlock
I guess this pull request is dropping a bit out of the sky, but I realized I needed the feature sooner rather than later so decided to just go for it. When it's merged it should fix #114 

Some refactoring was done to allow for integration register arrays, the most significant are
- expand function only takes 1 register instead of slice. This means that the function has to be called for every register that should be expanded. This was done so a register can be conditionaly expanded depending on if it's possible to convert to a Rust array or not.
- register_to_ru [... cut]
closed 2017-09-20 by homunkulus