rust-embedded/svd2rust #115: Very repetitious output when deriving enumeratedValues

roysmeding, 2017-06-11:
As I've said elsewhere, I'm working on adding `enumeratedValues` to the SVD for the MCU I'm using. Since there are a lot of fields that use the same enumerations, I'm reusing them using e.g. `<enumeratedValues derivedFrom="GPIOA.MODER.MODER15.MODE" />` in each field.

However, when inspecting the output, I noticed this results in a lot of duplicated code. Most of it seems to be due to the separate write proxies for each field, with each proxy defining all the variant methods, despite their implementation being the exact same each time.

This isn't a huge problem, and I understand why the separate write proxy objects are necessary (each field has a different offset), but it seems like there should be a less boilerplate-y way to accomplish the same thing? The first thing that comes to mind is generics with numerical arguments, but those aren't a part of Rust yet, unfortunately. I'm fairly new to Rust, though, so maybe someone has a better idea?

Some of this would be ameliorated by #44 but that wouldn't stop the duplication entirely.

japaric, 2017-06-20:
> The first thing that comes to mind is generics with numerical arguments

I don't think generics are required. The field offset could be stored as a field in the proxy object:

``` rust
struct CommonFieldW<'a> {
    w: &'a mut W,
    offset: u8,
}

// instead of FieldAW use CommonFieldW { w, offset: 0 }
// instead of FieldBW use CommonFieldW { w, offset: 1 }
// instead of FieldCW use CommonFieldW { w, offset: 2 }
// ...
```

burrbull, 2022-11-07:
Closing as implemented. Open new issue if you find it.

closed 2022-11-07 by burrbull