rust-embedded/svd2rust #116: Large generated `lib` file is unusable by development tools such as intellij-rust

nicholastmosher, 2017-06-12:
Due to the huge contiguous file size of the `lib.rs` generated output, intellij-rust (and possibly other development help tools) is unable to parse the library for intelligent completion support. I've been able to manually fix this by breaking top-level modules into their own files (you can see an example [here](https://github.com/nicholastmosher/lpc176x5x-rs)), but this is a tedious process that involves fixing scopes by inserting new `use` statements manually, etc.

I think it'd be helpful if svd2rust had an option to output top-level modules (or further, perhaps every module) to individual module files for easier navigation and completion.

Emilgardis, 2017-06-17:
This was previously how svd2rust worked. It was changed (presumably) because it meant you'd have to reparse the svd for each peripheral. I think however an utility tool for splitting large .rs files into multiple files could be something to look into.

djmcgill, 2017-09-13:
I wrote such a utility: https://crates.io/crates/form

kjetilkjeka, 2017-09-24:
@djmcgill the form utility is great! I just used it on [s32k144](https://github.com/kjetilkjeka/s32k144.rs) and it's great to be able to see the code in the github viewer. And also beeing able to use git diffs on an isolated part of the generated code is very usefull when verifying that the regenerated code still makes sense.

Currently in the [documentation](https://docs.rs/svd2rust/0.11.4/svd2rust/) ```svd2rust -i STM32F30x.svd | rustfmt | tee src/lib.rs``` is the recommended cmd. I think it would make sense to change it to include form.

djmcgill, 2017-09-24:
@kjetilkjeka I'm glad you like it! I would recommend holding off on making it "official" until form has reached at least 1.0, its certainly not stable yet.

idubrov, 2017-10-23:
@djmcgill I tried it on [stm32f103xx](https://github.com/japaric/stm32f103xx) -- works great, thanks a lot!

P.S. Just make it official, that will make it stable **really** fast (after you get and fix all those thousands of issues like "why does it uses `/Users/davidmcgillicuddy` directory by default :) )!

jonas-schievink, 2018-03-18:
Another annoyance caused by the huge `lib.rs` files is that it's basically impossible to view code via rustdoc's `[src]` links

burrbull, 2022-11-07:
Closing due to `form`

closed 2022-11-07 by burrbull