rust-embedded/svd2rust #119: Support for MSP430

pftbest, 2017-06-29:
It is possible to use the current release of svd2rust to generate device crate for msp430,
but you need to manually fix some things in generated code:
1) Remove default perfipherals (https://github.com/japaric/svd2rust/pull/117)
2) Replace all uses of `cortex_m` crate with `msp430`
3) Replace all `extern "C"` with `extern "msp430-interrupt"`
4) Add `#![feature(abi_msp430_interrupt)]` declaration

The proposal is to add `--arch=msp430` command line flag to svd2rust, that will affect the generated code to make this steps unnecessary.

You can find an SVD file and an example of generated code in [msp430g2553](https://github.com/pftbest/msp430g2553) device crate.

japaric, 2017-06-29:
Thanks for the info, @pftbest!

japaric, 2017-07-06:
fixed in #120 🎉

closed 2017-07-06 by japaric