rust-embedded/svd2rust #127: Add a check that fields of a register do not overlap and fit inside of a register.

pftbest, 2017-07-11:
While doing more testing for MSP430 I have found that dslite files I use for generating SVD have a lot of registers that have fields with wrong offset.

japaric, 2017-07-11:
> Add a check that fields of a register do not overlap

Is this really wrong or is this the case where a register has two meanings and thus two sets of fields depending on how the register is being used. See example in #16.

Currently svd2rust performs no checks on the SVD file, but there's a XML schema file that you can use with a XML linter (SVDs files are XML files). I don't think schemas can perform the computations required for this specific check though.

protomors, 2017-07-11:
@japaric  This check is really impossible with XML schema. So if it is not hard to implement it would be nice if svd2rust printed a warning in such cases (like it is already doing in case of overlapping registers). It will help to find errors in custom SVD files (field that does not fit inside a register is definitely an error).

pftbest, 2017-07-11:
No, it's not #16, it was really wrong

But the good news is dslite files are not broken, they just have `begin` field in xml that have smaller value than the `end` filed, and I didn't account for that in my tool. Now I fixed it and it's all good again

japaric, 2017-07-11:
> But the good news is dslite files are not broken

👍 

> they just have begin field in xml that has larger value than the end field

That sounds ... weird.

> I didn't account for that in my tool

Are you not using dslite2svd?

> it's all good again

Can we close this then?

pftbest, 2017-07-11:
> Are you not using dslite2svd?

no, i'm not, i wrote my own tool in rust, because i don't know ruby :/

> Can we close this then?

Well, I have solved my issue, but the point still stands, if svd2rust had checked the fields like it does with registers, I would have catched this bug a month ago

burrbull, 2022-11-07:
Closing as should be fixed in `svd-parser`

closed 2022-11-07 by burrbull