rust-embedded/svd2rust #128: [MSP430] The generated code is not as efficient as in C.

pftbest, 2017-07-11:
A very simple code that just sets some bits and clears some bits:

```C
    P1OUT |= RED_LED;
    P1OUT &= ~GRN_LED;
```

```
    port_1_2.p1out.modify(|_, w| w.p0().set_bit()
                                  .p6().clear_bit());
```

Will produce only 2 instructions in C
```
    c150:	d2 d3 21 00 	bis.b	#1,	&0x0021	;r3 As==01
    c154:	f2 f0 bf ff 	and.b	#65471,	&0x0021	;#0xffbf
    c158:	21 00 
```

But it has 4 instructions in Rust:
```
    c0e0:	7d 40 be ff 	mov.b	#65470,	r13	;#0xffbe
    c0e4:	5d f2 21 00 	and.b	&0x0021,r13	;0x0021
    c0e8:	5d d3       	bis.b	#1,	r13	;r3 As==01
    c0ea:	c2 4d 21 00 	mov.b	r13,	&0x0021	;
```

This is what clang emits in llvm-ir, maybe we can do something similar
```
  %3 = load volatile i8, i8* @P1OUT, align 1, !tbaa !1
  %4 = or i8 %3, 1
  store volatile i8 %4, i8* @P1OUT, align 1, !tbaa !1
  %5 = load volatile i8, i8* @P1OUT, align 1, !tbaa !1
  %6 = and i8 %5, -65
  store volatile i8 %6, i8* @P1OUT, align 1, !tbaa !1
```

pftbest, 2017-07-11:
Maybe we can have a special api to make writing such code easier, like
```Rust
port_1_2.p1out.set(|s| s.p0().p1().p2());
port_1_2.p1out.clear(|c| c.p6().p5().p4());
```

japaric, 2017-07-11:
> A very simple code that just sets some bits and clears some bits:

These two snippets are not equivalent. The C version performs *two* volatile (RMW) operations on the register. Whereas the Rust version performs only one.

> I can get the same behavior if I do the operations separately, but it's too verbose(

This is the semantically equivalent of the C version.

> Maybe we can have a special api to make writing such code easier, like

How big is the difference (# of instructions) between the Rust version and the C version of this snippet?

protomors, 2017-07-11:
Recently I had similar idea but for a different reason. Because on Cortex-M3 it is possible to write or read a single bit using bit-band region of memory. For example to clear interrupt flag I use
```
tim.sr.modify(|_, w| w.uif().bit(false));
```
and it compiles into
```
movs	r0, #16
movt	r0, #16384
ldr	r1, [r0, #0]
bic.w	r1, r1, #1
str	r1, [r0, #0]
```
Which is not too bad, but instead of ldr-bic-str it could be a siple str to special address. And as a bonus bit-banding a single bit is fully atomic (not important in this case). I just think that benefits from this not worth it to create new API like
```
tim.sr.uif(false);
```
or something similar. And very much will depend from LLVM optimizations. That is why I did not create an issue about it.

But this use case is very common. For example to switch STM32F103 to 72MHz clock I wrote
```
    rcc.cr.modify(|_, w| w.hseon().bit(true));
    while rcc.cr.read().hserdy().bit_is_clear() {};
    rcc.cfgr.write(|w| w.pllsrc().hse().pllmul().mul9().ppre1().div2());
    rcc.cr.modify(|_, w| w.pllon().bit(true));
    while rcc.cr.read().pllrdy().bit_is_clear() {};
    flash.acr.modify(|_, w| w.latency().two());
    rcc.cfgr.modify(|_, w| w.sw().pll());
    while !rcc.cfgr.read().sws().is_pll() {};
    rcc.cr.modify(|_, w| w.hsion().bit(false));
    while rcc.cr.read().hsion().bit() {};
```
Then the difference in the size of the code begins to add up. It would be nice if existing API optimized to more efficient code in case of single bit operations. But i don't know if this even possible.

japaric, 2017-07-11:
Bit-banding is not available on all Cortex-M devices. (I recall reading somewhere than new chips are not including it because it uses quite a bit of silicon area and is not that used in practice). If we want to provide an API for this it should only be available on devices that have the bit banding feature. I think SVD files don't include this information (whether bit banding is available or not).

>  And as a bonus bit-banding a single bit is fully atomic (not important in this case).

Just want to note that using two atomic operations don't make the whole operation atomic. If you mark each single atomic as safe that can potentially make data races safe.

protomors, 2017-07-11:
>Bit-banding is not available on all Cortex-M devices.

Yes. It is not available on Cortex-M0. And even on M3 and M4 it is an optional block (but I am not avare of any micros it these families that does not have it).

>Just want to note that using two atomic operations don't make the whole operation atomic.

This is why I wrote that it is so only for operations with single bits. And it is another reason against special API. Such low level optimizations are better to be handled by compiler (because they are very dependent of the target device).

pftbest, 2017-07-11:
> How big is the difference (# of instructions) between the Rust version and the C version of this snippet?

If you are asking about the last snippet, it has the same 2 instructions like the original code.
I was just pointing out that it's more code, and you have to manually remember that you always do `set_bit` in one modify, and `clear_bit` in another modify, or else you get a slow code.

closed 2017-07-12 by pftbest