rust-embedded/svd2rust #131: Get the address of a peripherals data register?

cs2dsb, 2017-07-28:
I've looked through the doc and the code generated from my svd and I can't find a way to get the address of a peripherals data register for DMA purposes.

For example on ADC1 there is a DR register which will contain the result of conversion.

To set up DMA conversions from the ADC I believe I'll need the address of this location in memory.

I've had a look through this repo, cortex-m and cortex-m-rtfm and I can't find any way of accomplishing this.

Do you have any thoughts on how/where this should be accomplished? If you give me a steer I'm happy to make the changes and submit a PR. My first guess was adding another method alongside read/write/modify/reset called address that returns the usize address of the register calculated from the peripheral location + register offset.

protomors, 2017-07-29:
You can look how crate blue-pill implements DMA API. I believe that what you need is in function "write_all" here https://github.com/japaric/blue-pill/blob/master/src/serial.rs
This is how it gets pointer to USART data register and puts it into DMA.
```
        dma1.cpar4
            .write(|w| unsafe { w.bits(&usart1.dr as *const _ as u32) });
```
And if you are using rtfm framework - then new version has new way of working with resources. According to blog post http://blog.japaric.io/rtfm-v2/ it also allows to use resources with DMA:

> If you were wondering “what’s up with the double dereference (**) in the toggle function?” that’s required becaused the type of r.ON is &mut Static<bool> instead of &mut bool; both are semantically equal because Static is just a newtype. The Static newtype comes in handy when dealing with DMA based APIs and code that deals with resources in a generic fashion.

But I didn't try out new version of rtfm yet. So I cannot provide example for it.

cs2dsb, 2017-07-29:
Looks like there is loads of useful stuff in blue-pill and rtfm v2, I'll have a look at that. I suspect it solves this problem so I'll close this issue :)

Thanks

closed 2017-07-29 by cs2dsb