rust-embedded/svd2rust #135: Clone, Copy traits for RegW

nikhilkalige, 2017-08-07:
I found that Clone and Copy traits are derived for R part of registers, but not for W part of the register enums. 
```rust
pub enum MSTRW {
    # [ doc = "`0`" ]
    SLAVE,
    # [ doc = "`1`" ]
    MASTER,
}

# [ derive ( Clone , Copy , Debug , PartialEq ) ]
pub enum MSTRR {
    # [ doc = "undocumented" ]
    SLAVE,
    # [ doc = "undocumented" ]
    MASTER,
}
```
I was hoping to make use of these information to use as follows
```rust
pub unsafe trait SPI: Deref<Target = i2s2ext::RegisterBlock> {
    type Ticks: Into<u32>;

    fn init(&self, role: i2s2ext::cr1::MSTRW);
}

pub struct Spi<'a, S, D>
    where S: Any + SPI,
          D: DMA + 'a
{
    pub reg: &'a S,
    role: i2s2ext::cr1::MSTRW,
    dmatx: &'a D,
}

impl<'a, S, D> Spi<'a, S, D>
    where S: Any + SPI,
          D: Any + DMA
{
    fn init(&mut self, role: Role) {
        self.role = role;
        self.reg.cr1.modify(|_, w| w.mstr().variant(self.role));
    }
}
```
This currently fails as `self.role` is copyable.

japaric, 2017-09-19:
> So any reason that they are not implemented?

If there was one I don't remember ... 😅

chrysn, 2018-06-22:
Please derive at least PartialEq for the W enums. Those would allow HALs to use them directly at configuration time and check before writing them.

closed 2019-02-23 by bors[bot]