rust-embedded/svd2rust #136: Panic on tree.get_child("mpuPresent")

wheelin, 2017-08-18:
Hi,

I'm beginning a project based on the ST [SensorTile](http://www.st.com/en/evaluation-tools/steval-stlcs01v1.html) module and have a little problem generating the rust cmsis library with svd2rust. Here is the panic message : 

```
thread 'main' panicked at '/home/greg/.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.5.2/src/lib.rs:148 tree.get_child("mpuPresent")', /checkout/src/libcore/option.rs:823
stack backtrace:
   0: std::sys::imp::backtrace::tracing::imp::unwind_backtrace
             at /checkout/src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
   1: std::sys_common::backtrace::_print
             at /checkout/src/libstd/sys_common/backtrace.rs:71
   2: std::panicking::default_hook::{{closure}}
             at /checkout/src/libstd/sys_common/backtrace.rs:60
             at /checkout/src/libstd/panicking.rs:355
   3: std::panicking::default_hook
             at /checkout/src/libstd/panicking.rs:371
   4: std::panicking::rust_panic_with_hook
             at /checkout/src/libstd/panicking.rs:549
   5: std::panicking::begin_panic
             at /checkout/src/libstd/panicking.rs:511
   6: std::panicking::begin_panic_fmt
             at /checkout/src/libstd/panicking.rs:495
   7: rust_begin_unwind
             at /checkout/src/libstd/panicking.rs:471
   8: core::panicking::panic_fmt
             at /checkout/src/libcore/panicking.rs:69
   9: core::option::expect_failed
             at /checkout/src/libcore/option.rs:823
  10: svd_parser::Device::parse
  11: svd_parser::parse
  12: svd2rust::main
  13: __rust_maybe_catch_panic
             at /checkout/src/libpanic_unwind/lib.rs:98
  14: std::rt::lang_start
             at /checkout/src/libstd/panicking.rs:433
             at /checkout/src/libstd/panic.rs:361
             at /checkout/src/libstd/rt.rs:59
  15: __libc_start_main
  16: _start
``` 

I'm quite new to Rust and can't know if the problem comes from the svd file or if it's an error coming from svd2rust. I've uploaded the sensorTile's mcu svd here --> [stm32l4x6](https://github.com/wheelin/RsGclock/blob/master/STM32L4x6.svd)

I hope I'm on the right place to discuss about this problem!

Thanks for your answers!

Greg

pftbest, 2017-08-18:
Looks like some fields are missing in the cpu section of your svd file. You can try to add them manually. There is an example in documentation:

https://siliconlabs.github.io/Gecko_SDK_Doc/CMSIS/SVD/html/group__cpu_section__gr.html

closed 2017-08-19 by wheelin

pftbest, 2017-08-19:
But did you manage to fix the issue? Maybe svd2rust needs a better error messages...

wheelin, 2017-08-20:
Yes! Has you said, some fields were missing in the svd file. I've added the mpu field and another error has shown up, I've added this field as well and then it was OK. 

Yes, a better error message would be great, just before the panic to announce that maybe some fields are missing.

japaric, 2017-08-22:
FWIW, these panics come from the *parser* not from svd2rust. Sadly the parser doesn't have proper error handling so we can improve error messages in svd2rust until that is fixed in the parser.