rust-embedded/svd2rust #138: creates long line with a lot of spaces at the top of output

x37v, 2017-08-31:
revision 72cc2db4f18466c4eeb90b6949b44e9784053aaf

`~/local/src/svd2rust/target/debug/svd2rust -i ./data/STM32F030.svd | rustfmt | tee src/lib.rs`

my first line was:
`+# ! [ cfg_attr ( feature = "rt" , feature ( global_asm ) ) ] # ! [ cfg_attr ( feature = "rt" , feature ( macro_reexport ) ) ] # ! [ cfg_attr ( feature = "rt" , feature ( used ) ) ] # ! [ doc = "Peripheral access API for STM32F030 microcontrollers (generated using svd2rust v0.11.3)\n\nYou can find an overview of the API [here].\n\n[here]: https://docs.rs/svd2rust/0.11.3/svd2rust/#peripheral-api" ] # ! [ deny ( missing_docs ) ] # ! [ deny ( warnings ) ] # ! [ allow ( non_camel_case_types ) ] # ! [ feature ( const_fn ) ] # ! [ no_std ]extern crate cortex_m ;`

which I had to hand edit to fix up

here is my mangled commit [the source data file is in the *data* directory]
https://gitlab.com/xnor/stm32f030/commit/e60e5c90558bf18a4c7e07ae8573d81e36cc4b03
then hand edited:
https://gitlab.com/xnor/stm32f030/commit/3cd906cd8e8b6633887b8477d1a7b97f24ff5661

Emilgardis, 2017-08-31:
Shouldn't this be filed to [rustfmt](https://github.com/rust-lang-nursery/rustfmt)?

kjetilkjeka, 2017-08-31:
At least for me, rustfmt outputs that this line is too long to format and gives up.

Wouldn't the practical solution be to but the directives on seperate lines so rustfmt will be able to format them?

japaric, 2017-08-31:
I agree with @Emilgardis; I think this should be filled in the rustfmt repo. (By the way, I can't repro this with rustfmt-nightly-0.2.4, svd2rust-0.11.3 and stm32f40x.svd).

@kjetilkjeka the `quote` crate we are using under the hood emits all the code in a single line, AFAIK, so I don't see a way to do that.

protomors, 2017-08-31:
I also had this problem. But looks like it is really fixed in latest version of rustfmt. You can see it by this diff
https://github.com/japaric/stm32f103xx/commit/64413522226be8f94cf218eebab1e1da77b81b35
Before file lib.rs started with one long line on which rustfmt gave up. But after this regeneration it is correctly split.

x37v, 2017-08-31:
yeah, I got errors form rustfmt indicating giving up on long lines, I have *0.9.0-nightly*
Either way, wouldn't it make sense for svd2rust to insert new lines before *#* and then it should work with the current rustfmt as well?

`# ! [ cfg_attr ( feature = "rt" , feature ( global_asm ) ) ] # ! [ cfg_attr ( feature = "rt" , feature ( linkage ) ) ] # ! [ cfg_attr ( feature = "rt" , feature ( macro_reexport ) ) ] # ! [ cfg_attr ( feature = "rt" , feature ( used ) ) ] # ! [ doc = "Peripheral access API for STM32F103XX microcontrollers (generated using svd2rust v0.11.2)\n\nYou can find an overview of the API [here].\n\n[here]: https://docs.rs/svd2rust/0.11.2/svd2rust/#peripheral-api" ] # ! [ deny ( missing_docs ) ] # ! [ deny ( warnings ) ] # ! [ allow ( non_camel_case_types ) ] # ! [ feature ( const_fn ) ] # ! [ no_std ]extern crate cortex_m ;`

though I guess the doc string wouldn't work either.
Here is the result of the newline before each *#* then piped through rustfmt
```
# ! [ cfg_attr ( feature = "rt" , feature ( global_asm ) ) ]
# ! [ cfg_attr ( feature = "rt" , feature ( linkage ) ) ]
# ! [ cfg_attr ( feature = "rt" , feature ( macro_reexport ) ) ]
# ! [ cfg_attr ( feature = "rt" , feature ( used ) ) ]
# ! [ doc = "Peripheral access API for STM32F103XX microcontrollers (generated using svd2rust v0.11.2)\n\nYou can find an overview of the API [here].\n\n[here]: https://docs.rs/svd2rust/0.11.2/svd2rust/
#peripheral-api" ]
# ! [ deny ( missing_docs ) ]
# ! [ deny ( warnings ) ]
# ! [ allow ( non_camel_case_types ) ]
# ! [ feature ( const_fn ) ]
# ! [ no_std ]extern crate cortex_m ;
```
current rustfmt doesn't seem to take out all the extra whitespace, or put these on 2 separate lines even if I add a new line before each *#*
`# ! [ no_std ]extern crate cortex_m ;`

Similarly, wouldn't it make sense to add a newline after each semicolon in the output?

pelrun, 2017-10-11:
Latest rustfmt from the "rustfmt" crate (0.9.0-nightly) doesn't work. Latest rustfmt from the "rustfmt-nightly" crate (0.2.8) does.

Emilgardis, 2018-03-02:
I'm closing this as this doesn't seem to be a problem anymore (rustfmt v0.4.0).

closed 2018-03-02 by Emilgardis