rust-embedded/svd2rust #145: svd2rust crashes with some SVDs

maximevince, 2017-09-18:
When trying
"svd2rust -i BlueNRG1.svd"
it crashes in: thread 'main' panicked at 'svd-parser-0.5.2/src/lib.rs:411 tree.get_child("bitWidth")', src/libcore/option.rs:819:4

BlueNRG1.svd can be found as part of "STMicroelectronics BlueNRG-1 Series Device Support" here: https://www.keil.com/dd2/pack/

Detegr, 2017-09-19:
I'm having similar issue with SVD's in http://www.st.com/resource/en/svd/stm32l0_svd.zip
```
$ svd2rust -i STM32L0x3.svd
tree.get_child("mpuPresent")', /checkout/src/libcore/option.rs:839:4
```

pftbest, 2017-09-19:
svd parser doesn't have a proper error handling, so it just panics when SVD file is incorrect. See https://github.com/japaric/svd2rust/issues/136

japaric, 2017-09-19:
@maximevince Hmm, that may be a bug in the SVD parser since the [SVD spec](https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_Format_pg.html) says that `<bitWidth>` is [optional](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html). in bit ranges.

@Detegr `<mpuPresent>` is a mandatory subfield of [`<cpu>`](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_cpu.html). Your SVD file doesn't adhere to the standard so you'll have to fix it.

Emilgardis, 2018-03-02:
BlueNRG1.svd has `derivedFrom` attributes on `MFT1_TIMER_B`, which is why svd-parser crashes. 
Related japaric/svd#21

robinvalk, 2019-04-30:
I'm having the same issue with the BlueNRG2.svd file. Is there anything I can edit manually to get this file parsed?

robinvalk, 2019-04-30:
Allright, I manually mapped all the bitWidth fields and fixed some issues in the SVD itself. File is now parsed.

I see that the derivedFrom attribute parsing is not yet implemented on the Field element. When I find the time I might create a PR to implement this.

robinvalk, 2019-05-06:
Unfortunately, I'm not going to find the time...

burrbull, 2022-11-07:
Should be fixed. Reopen if it is not.

closed 2022-11-07 by burrbull