rust-embedded/svd2rust #146: Conversion failing on XMC4100

lucasbrendel, 2017-09-24:
I am running svd2rust on an XMC4100.svd file and it is panicing with the following

> 'C:\Users\lucas\.cargo\registry\src\github.com-1ecc
6299db9ec823\svd-parser-0.5.2\src\lib.rs:576 parse::u32(t)', src\libcore\opti
on.rs:839:4

If I followed it correctly it is for a check on 'resetMask'.

I ran the SVD against an XML validator for the SVD schema and it stated everything was ok.

Any suggestions on what i could test to track down the issue?

kjetilkjeka, 2017-09-24:
Would you mind upload the .svd file or create a minimal example that reproduces this bug?

lucasbrendel, 2017-09-24:
Sorry about that. Should have done that earlier
[XMC4100.zip](https://github.com/japaric/svd2rust/files/1327450/XMC4100.zip)

jamesmunns, 2018-03-13:
@lucasbrendel could you please retry this with the current `master` branch? It looks like some related issues were cleaned up.

Thanks!

lucasbrendel, 2018-03-15:
@jamesmunns yeah I will take a look at it this weekend. Thanks for the update

jamesmunns, 2018-03-24:
@lucasbrendel any update?

lucasbrendel, 2018-05-05:
@jamesmunns Sorry for the long wait but yes, using master with svd-parser 0.6.0 fixes my issues. The rust file will at least be generated. I haven't checked anything past that yet. The latest release of svd2rust using svd-parser 0.5.2 does not work though, just an FYI

closed 2018-05-05 by lucasbrendel