rust-embedded/svd2rust #150: SVD Registers with no dimIndex

lucasbrendel, 2017-10-01:
According to the CMSIS standard:
https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_special.html 

dimElementGroups with "[%s]" in their name are not suppose to use the dimIndex tag. I have an SVD that currently is doing this but getting an error due to the tag missing. The SVD file is loaded in #146 but will only get that far if my PR in japaric/svd#40 is used.

Requesting the the dimIndex missing be allowed if name contains [%s]

therealprof, 2017-10-03:
The Atmel SAM SVD files also have that exact same problem.

lucasbrendel, 2017-10-05:
I am working on a fix for it. I have something that will get passed the issue and generate the .rs file but I am still testing to make sure it actually does what it needs to

commit e3722335 (2018-02-24): Merge #168

168: Handle missing dimIndex on registers r=japaric a=kevinmehall

This supersedes @lucasbrendel's similar PR #153 and fixes #150.

Per the [SVD spec](http://arm-software.github.io/CMSIS_5/SVD/html/elem_special.html#dimElementGroup_gr), dimIndex is not required:

> By default, \<dimIndex\> is a value starting with 0.
> Remark: Do not define \<dimIndex\> when using the placeholder [%s] in \<name\> or \<displayName\>.

I interpret this as the element being optional with "%s" and not allowed with "[%s]" (this change does not reject the latter). Atmel SAMD21 SVDs use "%s" with no dimIn [... cut]
PR #168 (2018-02-24): Handle missing dimIndex on registers
This supersedes @lucasbrendel's similar PR #153 and fixes #150.

Per the [SVD spec](http://arm-software.github.io/CMSIS_5/SVD/html/elem_special.html#dimElementGroup_gr), dimIndex is not required:

> By default, \<dimIndex\> is a value starting with 0.
> Remark: Do not define \<dimIndex\> when using the placeholder [%s] in \<name\> or \<displayName\>.

I interpret this as the element being optional with "%s" and not allowed with "[%s]" (this change does not reject the latter). Atmel SAMD21 SVDs use "%s" with no dimIndex for 0-indexed arrays.

This also tests [... cut]
closed 2018-02-24 by japaric