rust-embedded/svd2rust #154: Generated Code fails to compile due to missing modules

nigelb, 2017-10-21:
When I try to compile the code generated from this SVD file: [ATSAML21E18B.zip](https://github.com/japaric/svd2rust/files/1403763/ATSAML21E18B.zip)

With this procedure:

1. `cargo init`
2. `cargo add bare-metal vcell cortex-m`
3. `echo "max_width = 10000" > rustfmt.toml` to prevent the line length errors
4. `svd2rust -i ATSAML21E18B.svd | rustfmt | tee src/lib.rs`
5. `cargo build`

I get the following errors:

```
error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66240:21
      |
66240 |     register_block: sercom0::RegisterBlock,
      |                     ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66243:19
      |
66243 |     type Target = sercom0::RegisterBlock;
      |                   ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66244:25
      |
66244 |     fn deref(&self) -> &sercom0::RegisterBlock {
      |                         ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66252:21
      |
66252 |     register_block: sercom0::RegisterBlock,
      |                     ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66255:19
      |
66255 |     type Target = sercom0::RegisterBlock;
      |                   ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66256:25
      |
66256 |     fn deref(&self) -> &sercom0::RegisterBlock {
      |                         ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66264:21
      |
66264 |     register_block: sercom0::RegisterBlock,
      |                     ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66267:19
      |
66267 |     type Target = sercom0::RegisterBlock;
      |                   ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66268:25
      |
66268 |     fn deref(&self) -> &sercom0::RegisterBlock {
      |                         ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66276:21
      |
66276 |     register_block: sercom0::RegisterBlock,
      |                     ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66279:19
      |
66279 |     type Target = sercom0::RegisterBlock;
      |                   ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66280:25
      |
66280 |     fn deref(&self) -> &sercom0::RegisterBlock {
      |                         ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66288:21
      |
66288 |     register_block: sercom0::RegisterBlock,
      |                     ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66291:19
      |
66291 |     type Target = sercom0::RegisterBlock;
      |                   ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `sercom0`
     --> src/lib.rs:66292:25
      |
66292 |     fn deref(&self) -> &sercom0::RegisterBlock {
      |                         ^^^^^^^ Use of undeclared type or module `sercom0`

error[E0433]: failed to resolve. Use of undeclared type or module `tc0`
     --> src/lib.rs:79227:21
      |
79227 |     register_block: tc0::RegisterBlock,
      |                     ^^^ Use of undeclared type or module `tc0`

error[E0433]: failed to resolve. Use of undeclared type or module `tc0`
     --> src/lib.rs:79230:19
      |
79230 |     type Target = tc0::RegisterBlock;
      |                   ^^^ Use of undeclared type or module `tc0`

error[E0433]: failed to resolve. Use of undeclared type or module `tc0`
     --> src/lib.rs:79231:25
      |
79231 |     fn deref(&self) -> &tc0::RegisterBlock {
      |                         ^^^ Use of undeclared type or module `tc0`

error[E0433]: failed to resolve. Use of undeclared type or module `tc0`
     --> src/lib.rs:79239:21
      |
79239 |     register_block: tc0::RegisterBlock,
      |                     ^^^ Use of undeclared type or module `tc0`

error[E0433]: failed to resolve. Use of undeclared type or module `tc0`
     --> src/lib.rs:79242:19
      |
79242 |     type Target = tc0::RegisterBlock;
      |                   ^^^ Use of undeclared type or module `tc0`

error[E0433]: failed to resolve. Use of undeclared type or module `tc0`
     --> src/lib.rs:79243:25
      |
79243 |     fn deref(&self) -> &tc0::RegisterBlock {
      |                         ^^^ Use of undeclared type or module `tc0`

error[E0573]: expected type, found constant `CPUID`
     --> src/lib.rs:99994:37
      |
99994 |     #[doc = "CPUID"] pub CPUID: &'a CPUID,
      |                                     ^^^^^ not a type

error[E0573]: expected type, found constant `DCB`
     --> src/lib.rs:99995:33
      |
99995 |     #[doc = "DCB"] pub DCB: &'a DCB,
      |                                 ^^^ not a type

error[E0573]: expected type, found constant `DWT`
     --> src/lib.rs:99996:33
      |
99996 |     #[doc = "DWT"] pub DWT: &'a DWT,
      |                                 ^^^ not a type

error[E0573]: expected type, found constant `FPB`
     --> src/lib.rs:99997:33
      |
99997 |     #[doc = "FPB"] pub FPB: &'a FPB,
      |                                 ^^^ not a type

error[E0573]: expected type, found constant `FPU`
     --> src/lib.rs:99998:33
      |
99998 |     #[doc = "FPU"] pub FPU: &'a FPU,
      |                                 ^^^ not a type

error[E0573]: expected type, found constant `ITM`
     --> src/lib.rs:99999:33
      |
99999 |     #[doc = "ITM"] pub ITM: &'a ITM,
      |                                 ^^^ not a type

error[E0573]: expected type, found constant `MPU`
      --> src/lib.rs:100000:33
       |
100000 |     #[doc = "MPU"] pub MPU: &'a MPU,
       |                                 ^^^ not a type

error[E0573]: expected type, found constant `NVIC`
      --> src/lib.rs:100001:35
       |
100001 |     #[doc = "NVIC"] pub NVIC: &'a NVIC,
       |                                   ^^^^ not a type

error[E0573]: expected type, found constant `SCB`
      --> src/lib.rs:100002:33
       |
100002 |     #[doc = "SCB"] pub SCB: &'a SCB,
       |                                 ^^^ not a type

error[E0573]: expected type, found constant `SYST`
      --> src/lib.rs:100003:35
       |
100003 |     #[doc = "SYST"] pub SYST: &'a SYST,
       |                                   ^^^^ not a type

error[E0573]: expected type, found constant `TPIU`
      --> src/lib.rs:100004:35
       |
100004 |     #[doc = "TPIU"] pub TPIU: &'a TPIU,
       |                                   ^^^^ not a type

error: aborting due to 32 previous errors
```

I have tested the same procedure with [this](https://raw.githubusercontent.com/posborne/cmsis-svd/master/data/STMicro/STM32F30x.svd) SVD file without issue.

japaric, 2017-10-24:
That [sounds] like #107: missing `<cluster>` support.

[sounds]: https://github.com/japaric/svd2rust/blob/5cc57baf96ad6855685ff46adcd1326b756b4be1/ci/script.sh#L107

nigelb, 2017-11-17:
Yep, sure does?

jamesmunns, 2018-03-13:
@nigelb Could you please try this with the current `master`? Cluster support was added, but I know some ATSAMD chips still have issued due to other problems.

If so, lets figure out which other reported issues are blocking you.

nigelb, 2018-03-15:
Hi @jamesmunns I have the same issue as #191 :

```
$ svd2rust.exe -V
svd2rust 0.12.0 (631ab3e 2018-03-09)

$ svd2rust -i ATSAML21E18B.svd
WARNING Some(Ident("mode1")) overlaps with another register block at offset 0. Ignoring.
WARNING Some(Ident("mode2")) overlaps with another register block at offset 0. Ignoring.
error: Cluster USART has no determinable `size` field
caused by: Warning! overlap while calculating Register Size within a Cluster! Cluster contents may be incorrectly aligned!
note: run with `RUST_BACKTRACE=1` for a backtrace

```

Thanks,

Nigel

jamesmunns, 2018-03-15:
Thanks for confirming @nigelb, I'd say follow that issue as @wez is working on a fix for it. We can keep this open to track your specific chip. I will update the title to represent this

burrbull, 2022-11-07:
should be fixed

closed 2022-11-07 by burrbull