rust-embedded/svd2rust #155: Generated code doesn't compile for TM4C123GH6PM

SolidTux, 2017-11-13:
When I try to compile the generated code for the [TM4C123GH6PM](https://github.com/posborne/cmsis-svd/blob/master/data/TexasInstruments/TM4C123GH6PM.svd) it throws lots of errors complaining about duplicate definitions ([complete log](https://gist.github.com/anonymous/6f4430f225fe2651c692e31eff6b7628)):

```
[...]

error[E0592]: duplicate definitions with name `reset`
      --> src/lib.rs:163262:13
       |
163262 | /             pub fn reset(&self) {
163263 | |                 self.write(|w| w)
163264 | |             }
       | |_____________^ duplicate definitions for `reset`
...
163331 | /             pub fn reset(&self) {
163332 | |                 self.write(|w| w)
163333 | |             }
       | |_____________- other definition for `reset`

error: aborting due to 255 previous errors

error: Could not compile `tm4c123gh6p`.

```

I generated the file using:

```
svd2rust -i TM4C123GH6PM.svd | rustfmt > src/lib.rs
```

rustfmt complained about some lines which are too long, but I'm not sure, if that's important.

japaric, 2017-12-20:
This fails because the SVD file has two registers with the same name is several register blocks. For example:

``` xml
        <register>
          <name>MCS</name>
          <description>I2C Master Control/Status</description>
          <addressOffset>0x00000004</addressOffset>
          <fields>
            <!-- .. -->
          </fields>
        </register>
        <register>
          <name>MCS</name>
          <description>I2C Master Control/Status</description>
          <alternateGroup>I2C0_ALT</alternateGroup>
          <addressOffset>0x00000004</addressOffset>
          <fields>
            <!-- .. -->
          </fields>
        <register>
```

In general this is not allowed by the standard, but it may be OK in this case because the second register specifies `<alternateGroup>` information (I'm not 100% sure). In any case, svd2rust doesn't know how to handle `<alternateGroup>` information.

But it seems to me that the idea here is that for this register its bits have *some* meaning when the register is read but the bitfields have a different meaning when the register is written to. Unfortunately, the SVD file contains no information about this being the case. So, at best what we could generate here is a union with two variants: one per set of bitfields where both variants allow read/write access to the bit fields, which is wrong because the data sheet indicates that one set is read-only (used for reads) and the other is write-only (used for writes) -- however, the SVD file doesn't contain any information about that so we can't do any better.

SolidTux, 2017-12-20:
Maybe there is another (better) SVD file somewhere. Maybe there could be some kind of option how to deal with alternate groups (ignore, treat as read only ...)? Even ignoring would be better (for me at least) than the current status as it would compile and you could at least use the registers which are defined in a normal group.

whitequark, 2018-01-25:
Use my SVD files, generated from dslite definitions: https://github.com/m-labs/dslite2svd.

jamesmunns, 2018-03-13:
@SolidTux does @whitequark's new SVDs address your needs?

SolidTux, 2018-03-13:
Sorry for taking so long. I don't get the redefine errors anymore, but now I get

```
error[E0133]: call to unsafe function requires unsafe function or block
     --> src/lib.rs:69892:17
      |
69892 |                 self.bit(true)
      |                 ^^^^^^^^^^^^^^ call to unsafe function

error[E0133]: call to unsafe function requires unsafe function or block
     --> src/lib.rs:69896:17
      |
69896 |                 self.bit(false)
      |                 ^^^^^^^^^^^^^^^ call to unsafe function

error: aborting due to 2 previous errors

If you want more information on this error, try using "rustc --explain E0133"
error: Could not compile `tm4c123gh6p`.
```

It doesn't matter whether I use tm4c123x.xml or tm4c123x-vendor.xml (what is the difference between those two?). I updated the library dependencies to the most recent version.

whitequark, 2018-03-13:
You need svd2rust from master. The tm4c123x crate and repository already include generated Rust code, is there a reason you're rebuilding it?

SolidTux, 2018-03-13:
I somehow didn't notice that there is a crate for this series of microcontrollers, thank you! I will try to use that.

SolidTux, 2018-03-13:
And with the master version it works, so I thinks this issue is resolved.

closed 2018-03-13 by SolidTux