rust-embedded/svd2rust #159: Thoughts on supplementing svd with other sources (.h files) primarily for enumerated values?

cs2dsb, 2017-11-28:
I’ve been using a device crate generated by svd2rust for a project I’m working on and have found the lack of enumerated values defined in my svd quite annoying. It seems ST haven’t added them in most cases.

I’ve been adding them myself from the reference manual as I need them but it’s tedious work.

I was wondering about parsing .h files from the stm cmsis and possibly libopencm3 to get these enumerated values added to the svds.

The cmsis files (for example stm32f303ze.h) contain some in the form peripheral_register_field_enumname. If you discount enum names that are only digits you get a clean looking list. There are some obvious missing ones but it’s a lot better than the svd alone.

I looked around for community driven projects and found libopencm3 which seems to have a fairly complete set ([for example](https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/common/gpio_common_f234.h)).

Has anyone looked into extracting this info from existing libraries? Any thoughts on if it’s worth trying?

japaric, 2017-12-01:
👍 for creating tools that *add* information to SVD files from any other source.

👎 for adding such functionality (e.g. C header parser) to svd2rust, itself. I don't think you were proposing this, though.

I think someone wrote a (Python?) tool for making edition of SVD files easier (i.e. make it easier to add  enumeratedValues info) but don't recall any of the details. Perhaps @cr1901 or j_ey (forgot their GH username) know / remember about it?

jgouly, 2017-12-01:
Sorry, must be thinking of someone else. I've avoided SVD files so far.

brandonedens, 2017-12-02:
@japaric the Python script is discussed here:
https://github.com/hackndev/zinc/issues/327
maybe?

jamesmunns, 2018-03-19:
@cs2dsb have you had a look at [bindgen](https://github.com/rust-lang-nursery/rust-bindgen) before? You might be able to use a combo of `svd2rust` and `bindgen` to get all the info you need, and perhaps blend them together to make a unified API.

ryankurte, 2018-03-20:
This should be reasonably easy to implement in rust (if that's desired) once we land svd generation in [svd#46](https://github.com/japaric/svd/issues/46).