rust-embedded/svd2rust #160: Split generated code into separate files

limira, 2017-12-07:
I do not work on embedded systems. But recently I have a friend who start his project on STM32F407VG. He use C/C++. I found your [blog](http://blog.japaric.io/quickstart/) and I want to try it and give him some guide about Rust for him to use Rust instead of C/C++. I follow your guide until the stage that use this tool: svd2rust. The generated source code for STM32F407 is very big: about 10MB. Currently, IDE support for Rust is very limited. So I have to read the generated Rust source to understand it. But with such a big file, navigate through is very slow. Can we automatically split the generated code into separate files?

kjetilkjeka, 2017-12-07:
Duplicate of: https://github.com/japaric/svd2rust/issues/116?

Anyway, you can use the [form](https://crates.io/crates/form) tool to split the .rs file into seperate files.

[This](https://github.com/kjetilkjeka/s32k144.rs/tree/master/src) kind of directory structure will be produced.

limira, 2017-12-07:
Thanks @kjetilkjeka, it is exactly what I am looking for!

closed 2017-12-07 by limira