rust-embedded/svd2rust #166: [CI] `VENDOR=Freescale` build is timing out

japaric, 2018-01-15:
This is probably caused by the recent changes in Cargo defaults: Cargo now uses incremental compilation by default in dev builds. Because we are reusing the same Cargo project to dump the code generated from different SVD files the incremental compilation must be doing a lot of useless work diffing the new version of the crate source vs the old one, which was generated from a totally different SVD file.

Disabling incremental compilation should fix the timeout.

jamesmunns, 2018-03-13:
@japaric is this okay to close? I believe it has been addressed already.

japaric, 2018-03-13:
Yes, I disabled incremental compilation and that seemed to fix it.

closed 2018-03-13 by japaric