rust-embedded/svd2rust #169: Problem with registers with %s not being expanded.

tim-programmer, 2018-01-19:
I'm using the SVD for the LPC4357 which I retrieved from here: https://community.nxp.com/docs/DOC-334628. I had to make some edits of some enum names since the original file causes duplicate symbols.

The problem is with the SGPIO registers which have the form <name>SFSP0_%s</name> but the generated rust code doesn't have the additional index value of 0-1 (from dimIndex) that I'd expect.

```
    pub struct RegisterBlock {
        #[doc = "0x00 - Pin configuration register for pins P0"]
        pub sfsp0_: [SFSP0_; 2],
    }
    #[doc = "Pin configuration register for pins P0"]
    pub struct SFSP0_ {
        register: VolatileCell<u32>,
    }

```

I've attached a reduced version of the file the reproduces the problem.
[LPC43xx_43Sxx_reduce.zip](https://github.com/japaric/svd2rust/files/1647679/LPC43xx_43Sxx_reduce.zip)

Emilgardis, 2018-01-29:
This is because `svd2rust` "unrolls" the array/list. This is to make functions taking an `%s`ed register easier to write.

tim-programmer, 2018-02-08:
@Emilgardis Yes I understand what it's supposed to do, the issue is that it isn't replacing the %s with anything. I'd expect pub struct SFSP0_0 and pub struct SFSP0_1 blocks rather than just multiple pub struct SFSP0_ blocks.

Emilgardis, 2018-02-08:
This could be solved with having svd2rust make a trait `SFSP0_`, and then generate structs which implement that trait. The only problem with this approach is that we'd need to have an array like `[T;8] where T: SFSP0_`, which iirc is not possible today.

n8tlarsen, 2022-11-07:
This isn't really related to #662 but I may be able to offer some insight. In this case `svd2rust` is not expanding/unrolling the [dimElementGroup](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_special.html#dimElementGroup_gr) (identifiers with the %s notation) as suggested, but actually making an array using a common type. Since it uses a common type, that type has to have a name, which in most cases is best given by simply replacing the enumerating token `%s` with nothing. In your case, the trailing underscore is unfortunate and I might suggest that svd2rust replace it with something interesting like `n` to denote that the struct is shared by a `n`umber of `SFSP0_n` registers. What do @burrbull and @Emilgardis think?

burrbull, 2022-11-07:
Reopen if `keep_list` doesn't work.

closed 2022-11-07 by burrbull