rust-embedded/svd2rust #170: Problem with description containing ' ; '

thorhs, 2018-01-24:
When I was running against [MK66F18.svd](https://github.com/posborne/cmsis-svd/blob/master/data/Freescale/MK66F18.svd), I got strange errors when trying to compile.  Afer a quick look above the error, I noticed some incorrectly formatted doc strings.  It turns out that when the description tag in the input document contains ' ; ' (without the quotes), it doesn't format correctly.  I removed the leading space from the input document and the doc string got formatted correctly.

Example input line (56250) that caused issues:

```
<description>When DIFF=0:It is single-ended 12-bit conversion ; when DIFF=1, it is differential 13-bit conversion with 2&apos;s complement output.</description>
```

The error I get when trying to compile:
```
[rb747@rbf2803wv t1]$ xargo build --example hello
   Compiling t1 v0.1.0 (file:///Users/rb747/prog/rust/t1)
error: unknown start of token: `
      --> src/lib.rs:312007:58
       |
312007 |             #[doc = "Checks if the value of the field is `_00`"]
       |                                                          ^
       |
help: unicode character '`' (Grave Accent) looks like ''' (Single Quote), but it's not
      --> src/lib.rs:312007:58
       |
312007 |             #[doc = "Checks if the value of the field is `_00`"]
       |                                                          ^

error: Could not compile `t1`.

```

The malformed line (311983) in the code:
```
        pub enum MODER {; when DIFF=1, it is differential 11-bit conversion with 2's complement output" ] _10 , # [ doc = "When DIFF=0:It is single-ended 16-bit conversion..; when DIFF=1, it is differential 16-bit conversion with 2's complement output" ] _11}
```

thorhs, 2018-01-24:
On second thought, this may be happening in rustfmt.  When I pipe to a file without running rustfmt, it appears to be ok.

Emilgardis, 2018-03-02:
Issue seems to be fixed in 0.4.0 of rustfmt.

closed 2018-03-02 by Emilgardis