rust-embedded/svd2rust #172: When `isDefault` is present in an `enumeratedValue`, map all non-specified reads to that variant

jmesmon, 2018-01-26:
`isDefault` is _supposed_ to be an alternate to `value` within an `enumeratedValue`. Realistically, it's probably useful to allow both `value` and `isDefault` to be specified.

Current svd2rust appears to ignore it. It would likely be more correct to adjust the `_from` impls on the read proxies to translate all non-specified values into the enumeration marked with `isDefault`. Right now, it appears those will be mapped to a `_Reserved` enumeration.

jmesmon, 2018-01-26:
While this would be useful, it's not quite clear to me what the best way to allow both preserving the exact value written to the register & not require that the user picks 1 of several bit patterns with identical meaning.

xd009642, 2020-06-20:
I've just ran against this myself and was expecting the generated code to either give me an enum field without a value or be something akin to:

```Rust
pub enum FQC_A {
    Empty = 0,
    HasData(u8),
    Full = 16,
}
```

For context this is for a register describing the current capacity of a device FIFO which ranges from 0b00000->0b10000

PR #767 (2023-11-23): support of default enum value
Closes #172

This is usefull for (reading) fields like this:
[image]

For such fields changes look like:
[image]
[image]
closed 2023-11-23 by burrbull