rust-embedded/svd2rust #174: Handle SVD files which don't define register reset values

ecstatic-morse, 2018-02-07:
Atmel's ARM chips omit the `<resetValue>` element for register descriptions. `<resetValue>` is used to generate a `write` function which doesn't have to read the initial value from a register in order to preserve it while modifying individual fields.

Emilgardis, 2018-02-07:
You can access the bits via modify

kunerd, 2018-02-08:
I have used @ecstatic-morse's branch to generate the code for an ATtiny2313. The generated code compiles, but doesn't contain `.write()` and `.modify()` at all.

ecstatic-morse, 2018-02-08:
@kunerd, I was wrongly using "ocd-rw" to determine whether registers were read only, when really this field tells the on chip debugger not to read or write to this register because of side effects.

jamesmunns, 2018-03-13:
@kunerd @ecstatic-morse is this an issue with the ATDF to SVD parser, or with `svd2rust`?

I'd like to close this if there is nothing to be done at the `svd2rust` level.

kunerd, 2018-03-13:
@jamesmunns: Yes this an issue with ATDF to SVD parser, but closing depends on the questions: 

> Do we want to have a separate `atdf2rust` executable, or should `.atdf` support be added to `svd2rust` directly. And if we choose the separate executable way, should it be part of `svd2rust` or should it life in it's own repo?

I prefer the former one (for the first question), but I think the questions have not really been answered, yet. See #184

jamesmunns, 2018-03-13:
@kunerd is there anything that #184 doesn't cover that this one does? AFAICT, svd2rust does handle registers without reset values (even without `<resetValue>` elements), as long as they are marked as writable.

I will definitely follow #184 though (no opinions yet :) )

kunerd, 2018-03-13:
@jamesmunns Ahh, I missed the part with the `ocd-rw` above. So, I would say, this can be closed, but maybe you want to wait until @ecstatic-morse confirmed that, because he has done the implementation and I only tried to test it.

jamesmunns, 2018-03-24:
@ecstatic-morse any feedback?

japaric, 2018-03-27:
This should be supported. The SVD spec allows `<resetValue>` to be missing. In the case that resetValue is missing we should not generate the `reset` method and we should modify the `write` method somehow because that makes use of resetValue.

sjroe, 2018-09-26:
I've come across this recently as I'm trying to use an Atmel provided SVD file which doesn't have `<resetValue>` for all registers.

I belive that with no reset value the `reset` and `reset_value` methods should be omitted and the `write` method should instead initialise `w` to `0` as in:

```rust
let mut w = W { bits: 0 };
```

See https://github.com/sjroe/svd2rust/tree/resetvalue for an example implementation.

lambda, 2018-11-19:
I've pushed pull request #258 with a rebased version of #175 that also addresses the comments from @sjroe.

burrbull, 2022-11-07:
Closing due to `write_with_zero`. Reopen if you have better solution.

commit 476f735a (2018-11-27): Support registers without a defined reset value

In case no reset value is defined for a register, we suppress generation
of the `reset`, and `reset_value` functions.  For writable registers which
don't have a reset value, we pass the only possible value we could into the
closure, a zeroed bitfield.

Fixes: #174
closed 2022-11-07 by burrbull