rust-embedded/svd2rust #176: Generate API for "write 1 to toggle" registers

GGreenwood, 2018-02-18:
Hey all,

Over in rust-embedded/rfcs#40, user @kinnison  brought up that some of the STM32's USB control registers have a "write 0 to leave alone; write 1 to toggle" interface, and that svd2rust isn't generating the proper API for these registers. Since he hasn't filed an issue here yet, this is my attempt to document the feature.

In particular, the STM32F0x2 has a register for each USB endpoint with six of these toggleable bits. These registers are described in [RM0091](http://www.st.com/content/ccc/resource/technical/document/reference_manual/c2/f8/8a/f2/18/e6/43/96/DM00031936.pdf/files/DM00031936.pdf/jcr:content/translations/en.DM00031936.pdf) under the name `USB_EPnR`:

> When the application software writes ‘0, the value of DTOG_TX remains unchanged, while writing ‘1 makes the bit value toggle. This bit is read/write but it can only be toggled by writing 1.

[Here](https://raw.githubusercontent.com/GGreenwood/stm32f042/master/src/svd.rs) is a link to the current output generated by svd2rust v0.12.0 from [STM32F0x2.svd ](http://www.st.com/resource/en/svd/stm32f0_svd.zip)

**STM32f0x2.svd**
```
            <field>
              <name>DTOG_TX</name>
              <description>Data Toggle, for transmission
              transfers</description>
              <bitOffset>6</bitOffset>
              <bitWidth>1</bitWidth>
            </field>
```

**svd2rust API**
```
            #[doc = "Bit 6 - Data Toggle, for transmission transfers"]
            #[inline]
            pub fn dtog_tx(&self) -> DTOG_TXR {
                let bits = {
                    const MASK: bool = true;
                    const OFFSET: u8 = 6;
                    ((self.bits >> OFFSET) & MASK as u32) != 0
                };
                DTOG_TXR { bits }
            }
```

```
            #[doc = "Bit 6 - Data Toggle, for transmission transfers"]
            #[inline]
            pub fn dtog_tx(&mut self) -> _DTOG_TXW {
                _DTOG_TXW { w: self }
            }
```

Let me know if there's any other information I can dig up! I'm still learning embedded on Rust, and am really excited to see where things are headed.

Emilgardis, 2018-03-02:
To implement this we need to enable parsing of  `<modifiedWriteValues>` as [defined in the spec][elem_field].

To make it work you'd have to patch this element into the SVD aswell

```
            <field>
              <name>DTOG_TX</name>
              <description>Data Toggle, for transmission
              transfers</description>
              <modifiedWriteValues>oneToToggle</modifiedWriteValues>
              <bitOffset>6</bitOffset>
              <bitWidth>1</bitWidth>
            </field>
```

ah-, 2018-03-14:
Out of interest, did you get anywhere with this? Are you working on a USB driver?

GGreenwood, 2018-03-14:
@ah- I haven't gotten anywhere with this yet, since working on svd2rust would be above my current experience with Rust. I have plans to work with the STM32f0 USB peripherals in a month or two, which is why I brought this up.

If this begins to block my work on USB devices and I feel more comfortable with the language, I'd be glad to contribute or assist @Emilgardis, following the plan that they have laid out.

burrbull, 2022-11-07:
reopen if not fixed

closed 2022-11-07 by burrbull