rust-embedded/svd2rust #183: Improve `svd2rust`'s logging story

jamesmunns, 2018-02-28:
Currently there is only output on STDERR when some subset of failures occur. There is no feedback for items such as:

* Empty Registers
* Skipped Peripherals

Additionally, there are no different levels of logging, as of now all output is written directly to stderr.

We should consider adding:
* some kind of granular logging system, maybe using `slog` or `env_logger`
* additional error, warning, and info logs usable for development or troubleshooting

achan1989, 2018-09-23:
As a starter I can introduce a logging system, but keep the existing "error and warning messages get printed to stderr" behaviour.

What's the desired end goal?  Print info and higher to stderr by default, and use a `-v` command-line switch to enable debug-level log printing?

adamgreig, 2019-07-26:
#237 adds a logging system, so I'm closing this issue. If we need to improve logging further please open a subsequent issue.

closed 2019-07-26 by adamgreig