rust-embedded/svd2rust #187: [0.12.0] error: Register MR[%s] has no `dim_index` field

eupn, 2018-03-05:
Hello,

while trying to convert my QN908xC SVD file I got this error. Commit history says that exact or a similar error was fixed, but it seems that isn't the case.

```bash
svd2rust.exe -i qn908XC.svd
```

Output:
```
error: Register MR[%s] has no `dim_index` field
```

I also [attaching](https://gist.github.com/eupn/9b2b1be3623ba5a0e58118a4a47347d2) the SVD file that I've used.

jamesmunns, 2018-03-09:
Hey @eupn, Could you please try this with the current `master` branch? I think these issues have been fixed, but there hasn't been a release since then. I did check the SVD you provided, and it works for me. If it also works for you, this should just be waiting on a `0.12.1` or `0.13.0` release.

Emilgardis, 2018-03-09:
It is common for rust repos to update the version number on master with the release, instead of after. That is why the fix is on master but not released to 0.12.0

eupn, 2018-03-09:
Thanks guys for a reply, I definitely saw that fix landed on the master branch and I was suspicious about that fix somehow wasn't there in `0.12.0`.

closed 2018-03-09 by eupn