rust-embedded/svd2rust #194: Generated rust file contains an unreachable pattern

jscarrott, 2018-03-19:
When using svd2rust on the [nRF52840.svd](https://github.com/NordicSemiconductor/Nordic-Thingy52-FW/blob/master/external/sdk13/svd/nrf52840.svd) file the generated code contains an unreachable pattern and won't build without adding allow(unreachable_patterns) to the library.

`error: unreachable pattern
     --> /home/jscarrott/Git/nrf52840/src/lib.rs:29983:21
      |
29983 |                     216 => TXPOWERR::NEG40DBM,
      |                     ^^^
      |
note: lint level defined here
     --> /home/jscarrott/Git/nrf52840/src/lib.rs:1:487
`
I've ended up just adding the allow for now and using it anyway.

jamesmunns, 2018-03-19:
Hey @jscarrott - I think that this is because they have an old aliased name in the SVD. I had a similar problem with my bindings for the nrf52832: https://github.com/jamesmunns/nrf52/commit/f27102a564fe946125fdc3abf0460efb4d457bf9

My approach was to modify the SVD file, since it is not possible to have the two match statements resolve bidirectionally.

jscarrott, 2018-03-19:
Yeah, I'm sure that's fine. I'm just tinkering around with the board for fun and to teach myself about embedded rust and with the very vague goal of getting it to transmit some MQTT stuff over Bluetooth.

I'm going to take the opportunity to thank you for the great documentation on embedded rust with svd2rust. It's been a real learning curve getting some LEDs to turn on over the weekend but with the documentation on svd2rust and @japaric's blog, the answers were never too far away. The next thing to work out though is why the LEDs only turn on after I reset the board and not when my code runs....

jamesmunns, 2018-03-19:
I can't take any credit for the docs, but if you want to chat about Rust, the nRF family, and misc IoT stuff, feel free to email me (its in my profile), or catch up with us on `#rust-embedded` at irc.mozilla.org!

I'm going to change the title to be a little more broad, we may still decide to "gracefully" handle this issue in the future.

adamgreig, 2019-07-26:
I believe this is fixed by new svd-parser in #317, please re-open if the issue persists.

closed 2019-07-26 by adamgreig