rust-embedded/svd2rust #198: Why does register.write require returning the reference to the writer

Nemo157, 2018-03-26:
For reference

```rust
pub struct REGISTER { ... };
pub struct W { ... };
pub struct _FIELDW<'a> { ... };

impl REGISTER {
    fn write<F: FnOnce(&mut W) -> &mut W>(&self, f: F) { ... }
}

impl W {
     fn field(&mut self) -> _FIELDW { ... }
}

impl _FIELDW<'a> {
    fn value(self) -> &'a mut W { ... }
}
```

The only upside I see to this formulation is statically avoiding the user writing something like `register.write(|w| w.field());` which will not do anything since the value isn't being specified. The alternative way to avoid this is to specify `#[must_use("You need to specify a value for the field")]` on `_FIELDW` to give the user a lint error if they don't use it.

An alternative I'd prefer would be to allow passing values back from the `write` function, e.g.

```rust
impl REGISTER {
    fn write<R, F: FnOnce(&mut W) -> R>(&self, f: F) -> R { ... }
}
```

This can be useful because it allows nicely encapsulating multi-layered typestates to statically ensure the register is configured correctly. Currently to write something like this you need to use an option at the top level to write the returned typestate into since you can't guarantee that `write` will call your supplied callback.

```rust
trait InputMode {
    fn apply(w: &mut pin_cnf::W) -> Self;
}

trait PinMode {
    fn apply(w: &mut pin_cnf::W) -> Self;
}

struct Floating;

struct Input<M: InputMode> {
    mode: M,
}

struct Pin<'a, M: PinMode> {
    gpio: &'a GPIO,
    mode: M,
}

impl InputMode for Floating {
    fn apply(w: &mut pin_cnf::W) -> Self {
        w.pull().disabled();
        Floating
    }
}

impl<M: InputMode> PinMode for Input<M>
    fn apply(w: &mut pin_cnf::W) -> Self {
        w.dir().input().input().connect();
        let mode = M::apply(w);
        Input { mode }
    }
}

impl<'a, M: PinMode> Pin<'a, M> {
    fn set_mode<N: PinMode>(self) -> Pin<'a, N> {
        let mode = self.gpio.pin_cnf.write(N::apply);
        Pin { gpio: self.gpio, mode }
    }
}
```

Nemo157, 2018-03-26:
Thinking about it a little more a user can still write

```rust
register.write(|w| {
    w.field();
    w
});
```

currently, and get no errors or warnings that the field will not be changed.

burrbull, 2022-11-07:
marked with must_use. reopen if it still is problem

closed 2022-11-07 by burrbull