rust-embedded/svd2rust #204: use `riscv-rt` version 0.2.0

Emilgardis, 2018-04-27:
This will allow CI to run with the new compiler builtins change to `#[no_std]` crates

related: riscv-rust/riscv-rt#3, riscv-rust/riscv-rt#4

Emilgardis, 2018-04-27:
This is also needed for all crates that uses riscv-rt

japaric, 2018-05-06:
riscv-rt v0.2.0 is still broken. It misuses the `#[used]` macro. cc @dvc94ch 

I'll disable the riscv tests for now.

> This is also needed for all crates that uses riscv-rt, so we need to bump the version in code gen as well

svd2rust doesn't generate a Cargo.toml for device crates. The only place where the riscv-rt version appears is our CI and the svd2rust documentation.

dvc94ch, 2018-05-06:
@japaric sounds reasonable to me. I'm currently working on another project and I'm really bad at focusing on more than one thing.

Disasm, 2019-02-06:
@japaric Now both riscv and riscv-rt can be built against stable. Maybe we can enable riscv tests? I can add a PR for this.

japaric, 2019-02-06:
@Disasm sounds good to me. Please also update the crate level documentation (it mentions versions v0.3.x of riscv{,-rt})

Disasm, 2019-02-06:
@japaric done: https://github.com/rust-embedded/svd2rust/pull/279

closed 2019-02-07 by therealprof